shajen/rtl-sdr-scanner-cpp #2: How to squelch / sensitivity?

dandud100, 2022-02-17:
For me it just records noise no matter if i configure 0.0 or 40.2 gain in the config.

Here are three .mp3 for example but i got 100 in like 15 minutes. It is a zip as i can't upload .mp3
[rtl-sdr-scanner.zip](https://github.com/shajen/rtl-sdr-scanner-cpp/files/8090730/rtl-sdr-scanner.zip)

shajen, 2022-02-17:
In your case frequency `438.250 MHz` is probably in use by something. Many different devices (e.g. temperature sensor, camera, garage door) uses it. Please add frequency to ignore in config.json file and try again. There is not squelch to define, rtl-sdr-scanner do it automatically.

Try add this to config:
```
"ignored_frequencies_ranges": [
    {
      "start": 438200000,
      "stop": 438300000
    },
]
```

dandud100, 2022-02-17:
> In your case frequency `438.250 MHz` is probably in use by something. Many different devices (e.g. temperature sensor, camera, garage door) uses it. Please add frequency to ignore in config.json [...]

Yeah i see that now that you fixed #1 

But it is still not fully functional, see following Screenshots
[image]
[image]
[image]

shajen, 2022-02-17:
Interesting. Can you show your `config.json` file?

dandud100, 2022-02-17:
> Interesting. Can you show your `config.json` file? Please test on the latest version.

I use the latest.

My config.json

`{
  "scanner_frequencies_ranges": [
    {
      "start": 433075000,
      "stop": 434775000,
      "step": 25000
    },
    {
      "start": 446000000,
      "stop": 446200000,
      "step": 125
    },
    {
      "start": 456840000,
      "stop": 457020000,
      "step": 125
    },
    {
      "start": 457220000,
      "stop": 457400000,
      "step": 125
    },
    {
      "start": 467220000,
      "stop": 467400000,
      "step": 125
    }
  ],
  "ignored_frequencies_ranges": [
	{
      "start": 434400000,
      "stop": 434400000
    }
  ],
  "device": {
    "ppm_error": 0,
    "tuner_gain": 40.2,
    "bandwidth": 2000000
  },
  "recording": {
    "range_scanning_time_ms": 100,
    "max_silence_time_ms": 2000,
    "min_recording_time_ms": 1000,
    "sample_rate": 48000,
    "threads": 4,
    "enabled": true
  },
  "output": {
    "recordings": "sdr/recordings",
    "logs": "sdr/logs",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "server": {
    "address": "0.0.0.0",
    "port": 9999,
    "key": "",
    "threads": 4
  }
}
`

shajen, 2022-02-17:
I tested it and your config works ok with my rtl-sdr scanner on latest version. The frequencies on your screenshots are different that you posted in config.

dandud100, 2022-02-17:
> I tested it and your config works ok with my rtl-sdr scanner on latest version. The frequencies on your screenshots are different that you posted in config.

Yeah, you are right, i just found this out in the second you answered. It just does not use my config, if i run it manually it does but with my systemctl service it does not. What is wrong?

```
[Unit]
Description=rtl-sdr-scanner

[Service]
User=pi
WorkingDirectory=/home/pi/
Type=exec
ExecStart=/home/pi/rtl-sdr-scanner-cpp/build/auto-sdr config.json
Restart=always
RestartSec=10s
OOMScoreAdjust=-500

[Install]
WantedBy=multi-user.target

```

shajen, 2022-02-17:
Pass full path to `config.json`. Probably the path should be like this:
```
ExecStart=/home/pi/rtl-sdr-scanner-cpp/build/auto-sdr /home/pi/rtl-sdr-scanner-cpp/config.json
```

shajen, 2022-02-17:
It will work slowly, because every switch between frequencies range takes about ~100 ms. So you should definitely choose as few frequency ranges as possible to speed it up. 

But if the scanner finds an active frequency, it records it without switching as long as the signal is active.

dandud100, 2022-02-18:
> It will work slowly, because every switch between frequencies range takes about ~100 ms. So you should definitely choose as few frequency ranges as possible to speed it up.
> 
> But if the scanner finds an active frequency, it records it without switching as long as the signal is active.

Hmm, now i try this with only one frequency but why it does not record it?

[image]

Ta-2om, 2022-03-10:
and the squelch question?

Tried to set max recording time. But no change.

 "recording": {
    "range_scanning_time_ms": 100,
    "max_silence_time_ms": 4000,
    **"max_recording_time_s": 30,**
    "min_recording_time_ms": 3000,
    "sample_rate": 48000,
    "threads": 4,
    "enabled": true
    
    Thank you

neonmaus, 2023-01-22:
> ignored_frequencies_ranges

Is this function no longer included in the current version?

shajen, 2023-02-08:
@neonmaus 

Yes, update to latest version and see below:
https://github.com/shajen/rtl-sdr-scanner-cpp#ignored-frequencies

neonmaus, 2023-02-08:
> @neonmaus
> 
> Yes, update to latest version and see below: https://github.com/shajen/rtl-sdr-scanner-cpp#ignored-frequencies

Yeah! thanks for putting it back in :)

ZL1LAC, 2023-03-29:
Sorry to be off topic but that's a very interesting Tetra signal.

closed 2023-11-13 by shajen