shajen/rtl-sdr-scanner-cpp #5: Sending frequency, time and file name to SQL database

GarryCZ, 2022-06-21:
Hi, your project is amazing. It works absolutely great.
But I would have a query, I would need to write data (frequency, file name and time) to the SQL database (for example via http request)
Could you please advise me?
Thank you so much

shajen, 2022-06-23:
Hi,
good place to do this is `Recorder::start`. You should replace line
```c++
m_mp3Writer = std::make_unique<Mp3Writer>(m_config, frequency, mp3SampleRate);
```
with
```c++
if (m_mp3Writer) {
  addToDb(m_mp3Writer->getFrequency(), m_mp3Writer->getFilename(), m_mp3Writer->getTime());
}
m_mp3Writer = std::make_unique<Mp3Writer>(m_config, frequency, mp3SampleRate);
```
You should implement `addToDb` method with any `db`/`http` library you want and add `getFrequency`, `getFilename`, `getTime` getter to `Mp3Writer` class.

Another way to do this is add similar code in `Mp3Writer::~Mp3Writer()`.

GarryCZ, 2022-06-23:
thank you so much, i will try.

GarryCZ, 2022-06-24:
hi if i write it like this:

	if (m_mp3Writer) {
		cout << "sending data!" << endl;
		system("curl http://0.0.0.0:3000/test.php?RX_freq=");
	}
so if is not called at all.
if I put it outside of if, it is called when the broadcast is active.

Could you please advise me? thank you very much

GarryCZ, 2022-06-24:
hi, i wrote it in mp3_writer and it already works for me.
thank you very much

closed 2022-06-24 by GarryCZ

D0han, 2022-09-09:
@GarryCZ will you publish that code? Create PR maybe?