shajen/rtl-sdr-scanner-cpp #7: Multiple dongles

nfacha, 2022-09-05:
Hi :)
Assuming I have multiple SDR dongles attached, how do I select which one this will run on?

Nuno

@shajen

shajen, 2022-12-22:
I have merged already raw to master. It's ready to work. There is many many new feature. Please check master branch.

New config support multiple dongles based on device serial.

closed 2022-12-22 by shajen

dubrsl, 2022-12-23:
Some devices have the same serial and stop working if change it.

shajen, 2022-12-23:
Yeah I know that all `rtl-sdr` dongles from China have the same serial but you can change it.

Attach **just one** dongle and type:
```
rtl_eeprom -s 11111111
```
Then attach **next one** and type:
```
rtl_eeprom -s 22222222
```
Then you are able to configure each one separately in `sdr-scanner` config. Do not forget to replug dongle after change serial.

dubrsl, 2022-12-23:
Thank you!
I know about rtl_eeprom -s
But other software require specific serial and stop working after changes. So I can't set other serial. :(
It seems to me that it makes sense to add Bus, Port and Dev to the serial number for unique identification. Look in /sys/bus/usb/devices/

BTW,
@shajen , do you have other channel for communication? forum, slack, telegram anything else? Not convenient to communicate in issues. I have some ideas and FR for discuss.

shajen, 2022-12-27:
Adding support for finding devices via /sys/bus should be easy because both libraries support listing devices via /sys/bus.

Try to find me on discord @shajen.