shajen/rtl-sdr-scanner-cpp #8: cmake -B build -DCMAKE_BUILD_TYPE=Release

graememilne, 2022-11-27:
So i have followed the the instructions .cd into rtl-sdr-cpp folder $cmake -B build -DCMAKE_BUILD_TYPE=Release.
On doing this i get an error
[image]
Have googled for any help but all i get is confusing info .
This project looks great apart from the fustrating holdup .
Hope someone can come up with a simple remedy i can follow.
Thankyou and regards
graeme

shajen, 2022-11-28:
You need to install `nlohmann-json`. If your distro is Ubuntu or Debian type in terminal:
```
sudo apt-get install nlohmann-json3-dev
```
or find out how to install it on your distro if use another.

shajen, 2022-12-22:
I have merged already raw to master. It's ready to work. There is many many new feature. Please check master branch.

closed 2022-12-22 by shajen