shajen/rtl-sdr-scanner-cpp #10: Hackrf not working correctly, and some error.

Klammraute, 2022-12-25:
i try to monitor 3 frequencyranges but can not use a sample rate above 2mhz (hackrf can have a sample rate up to 20Mhz)
when i run the scanner and do a test sending with a radio 500m away its will be not recorded. if i have only 1 frequencyrange with 2mhz bandwith (eg: 144-146) it works but when i have a wider frequency range like 118-130 or 146-173 it record nothing. 

Second: in the Readme you say for starting run: "./build/auto-sdr config.json" but its "auto_sdr"
Question can i also add ignored frequencies like in the old version?

my config: 
{
  "scanner_frequencies_ranges": [
    {
      "device_serial": "auto",
      "ranges": [
        {
          "start": 146010000,
          "stop": 173010000,
          "step": 1000,
          "sample_rate": 2048000
        },
        {
          "start": 58000000,
          "stop": 87000000,
          "step": 1000,
          "sample_rate": 2048000
        },
        {
          "start": 446000000,
          "stop": 446200000,
          "step": 1000,
          "sample_rate": 2048000
        }
      ]
    }
  ],
  "devices": {
    "hack_rf": {
      "lna_gain": 16,
      "vga_gain": 42,
      "offset": 0
    }
  },
  "recording": {
    "max_noise_time_ms": 50,
    "min_time_ms": 2000,
    "min_sample_rate": 16000,
    "max_concurrent_recordings": 10
  },
  "detection": {
    "frequency_grouping_size": 10000,
    "frequency_range_scanning_time_ms": 64,
    "noise_learning_time_seconds": 30,
    "noise_detection_margin": 20,
    "torn_transmission_learning_time_seconds": 60
  },
  "output": {
    "logs": "sdr/logs",
    "recordings": "sdr/recordings",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "mqtt": {
    "hostname": "sdr-broker",
    "port": 1883,
    "username": "admin",
    "password": "password"
  }
}

shajen, 2022-12-26:
Make sure that the sample rate and step are correct. Follow the manual here https://github.com/shajen/rtl-sdr-scanner-cpp#custom-sample-rate-step-and-frequency-range Sample rate 20 MHz should work fine with HackRf but with steps 625, 1250, 2500 etc.

A single scan with 2 MHz sample rate takes about 100 ms + about 100 ms to change frequency. So if you set the frequency range 146-173 with sample rate 2 Mhz full scan take about ~2800 ms. So it's very probable that if the transmission is short it would not be recorder by scanner.

In the new version, there is no ignored frequency feature because sdr scanner automatically learns bussy frequencies and skips them.

Klammraute, 2022-12-26:
ok, that worked for a short time, but when i run the software a second time i got this error, how to fix?

sdr-scanner-sdr-monitor-setup-1   | Running migrations:
sdr-scanner-sdr-monitor-setup-1   |   No migrations to apply.
sdr-scanner-sdr-monitor-setup-1   | CommandError: Error: That username is already taken.

shajen, 2022-12-27:
It doesn't matter, ignore it. Db setup tries to add admin account, but is
already exists.

shajen, 2023-01-02:
I found that there is some problem with `hackrf` and `sample rate` above `10 MHz`. I'm working on it. I don't know if problem is with my device or with scanner software. Be patient and wait for update. For now it works ok with `sample rate` `2 Mhz`.