shajen/rtl-sdr-scanner-cpp #13: I can't set the frequency range

GarryCZ, 2022-12-29:
Hi , I can't set the frequency range with one SDR, 
it still says samples not ok, start: 144.000 kHz, stop: 146.000 kHz, step: 001.000 Hz, bandwidth: 002.048 kHz
samples not ok, start: 440.000 kHz, stop: 442.000 kHz, step: 001.000 Hz, bandwidth: 002.048 kHz
In the previous version of the scanner it worked perfectly.
Here is my config:
{
  "scanner_frequencies_ranges": [
    {
      "device_serial": "auto",
      "ranges": [
        {
          "start": 144000000,
          "stop": 146000000,
          "step": 1000,
          "sample_rate": 2048000
        },
        {
          "start": 440000000,
          "stop": 442000000,
          "step": 1000,
          "sample_rate": 2048000
        }
      ]
    }
  ],
  "devices": {
    "rtl_sdr": {
      "ppm_error": 40,
      "tuner_gain": 10.0,
      "offset": 0
    },
    "hack_rf": {
      "lna_gain": 0,
      "vga_gain": 0,
      "offset": 0
    }
  },
  "recording": {
    "max_noise_time_ms": 500,
    "min_time_ms": 2000,
    "min_sample_rate": 16000,
    "max_concurrent_recordings": 10
  },
  "detection": {
    "frequency_grouping_size": 10000,
    "frequency_range_scanning_time_ms": 64,
    "noise_learning_time_seconds": 30,
    "noise_detection_margin": 20,
    "torn_transmission_learning_time_seconds": 60
  },
  "output": {
    "logs": "sdr/logs",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "mqtt": {
    "hostname": "sdr-broker",
    "port": 1883,
    "username": "admin",
    "password": "password"
  }
}

shajen, 2023-01-02:
It was a bug, thanks for find it. I hot fixed it and push changes to master with some additional features. Please update. Keep in mind that there is some changes in `config.json`, please update it.

GarryCZ, 2023-01-03:
Hi, after the update I can't connect to SDR. 

Attaching to sdr-sdr-broker-1, sdr-sdr-monitor-reader-1, sdr-sdr-monitor-server-1, sdr-sdr-monitor-setup-1, sdr-sdr-scanner-1
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.809] [auto_sdr] [info]     [main]        start thread id: 8
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.809] [auto_sdr] [info]     [main]        start app auto_sdr
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.809] [auto_sdr] [info]     [main]        build type: release
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.809] [auto_sdr] [info]     [Mqtt]        start thread id: 9
sdr-sdr-broker-1          | 1672766707: New connection from 172.18.0.4:47408 on port 1883.
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.813] [auto_sdr] [info]     [Mqtt]        connected
sdr-sdr-broker-1          | 1672766707: New client connected from 172.18.0.4:47408 as auto-CBFD1FD5-45CB-0530-E31B-2052BCA20ABA (p2, c1, k60, u'admin').
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.975] [auto_sdr] [info]     [RtlSdr]      open device, index: 0, name: Generic RTL2832U OEM, manufacturer: DOMCAD_RX1, product: RTL2838UHIDIR, serial: 00000001
sdr-sdr-scanner-1         | 
sdr-sdr-scanner-1         | Kernel driver is active, or device is claimed by second instance of librtlsdr.
sdr-sdr-scanner-1         | In the first case, please either detach or blacklist the kernel module
sdr-sdr-scanner-1         | (dvb_usb_rtl28xxu), or enable automatic detaching at compile time.
sdr-sdr-scanner-1         | 
sdr-sdr-scanner-1         | usb_claim_interface error -6
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.980] [auto_sdr] [info]     [Mqtt]        stop thread id: 9
sdr-sdr-broker-1          | 1672766707: Client auto-CBFD1FD5-45CB-0530-E31B-2052BCA20ABA disconnected.
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.981] [auto_sdr] [error]    [main]        main exception: can not open rtl sdr device
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.981] [auto_sdr] [info]     [main]        stop app auto_sdr
sdr-sdr-scanner-1         | [2023-01-03 18:25:07.981] [auto_sdr] [info]     [main]        stop thread id: 8
sdr-sdr-scanner-1 exited with code 0
sdr-sdr-monitor-setup-1   | Operations to perform:
sdr-sdr-monitor-setup-1   |   Apply all migrations: admin, auth, contenttypes, graphs, logs, sdr, sessions
sdr-sdr-monitor-setup-1   | Running migrations:
sdr-sdr-monitor-setup-1   |   No migrations to apply.
sdr-sdr-broker-1          | 1672766726: New connection from 172.18.0.3:54017 on port 1883.
sdr-sdr-broker-1          | 1672766726: New client connected from 172.18.0.3:54017 as auto-1A3B7EB0-C76E-4913-8460-4BE78149B905 (p2, c1, k60, u'admin').
sdr-sdr-monitor-setup-1   | CommandError: Error: That username is already taken.
sdr-sdr-monitor-setup-1 exited with code 0

Interestingly, everything is fine when using rtl_test:
rtl_test
Found 1 device(s):
  0:  DOMCAD_RX1, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 100 bytes
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 6
Reattached kernel driver

thank you for your help

GarryCZ, 2023-01-03:
Sorry, the previous error is no longer valid I solved it by adding blacklist dvb_usb_rtl28xxu to /etc/modprobe.d/blacklist.conf.

now everything is running perfectly even with multiple bands.

Thank you so much, you're doing a great job

closed 2023-01-03 by GarryCZ