shajen/rtl-sdr-scanner-cpp #14: Script stops working after a new recording

GarryCZ, 2023-01-04:
Hi, sript always runs without problem, spectograms are saved, everything runs as it should until a new record is loaded.
then the whole program stops working completely. I'm attaching the output from the terminal.

`sdr-sdr-monitor-reader-1  | [2023-01-03 19:14:59.529][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:14:59.851][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:49.699 - 2023-01-03 19:14:58.130, 448000000 - 450000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:14:59.855][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:14:59.869][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.360][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:58.302 - 2023-01-03 19:15:06.641, 171000000 - 173000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.365][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.378][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.521][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:58.472 - 2023-01-03 19:15:06.820, 442000000 - 444000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.526][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.537][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.855][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:58.816 - 2023-01-03 19:15:07.168, 446000000 - 448000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.860][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:08.873][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.520][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:59.506 - 2023-01-03 19:15:07.851, 444000000 - 446000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.525][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.535][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.853][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:14:59.846 - 2023-01-03 19:15:08.190, 448000000 - 450000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.858][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:09.880][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.470][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:15:08.355 - 2023-01-03 19:15:16.783, 171000000 - 173000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.475][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.489][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.635][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:15:08.516 - 2023-01-03 19:15:16.950, 442000000 - 444000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.639][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.651][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-scanner-1         | [2023-01-03 19:15:18.802] [auto_sdr] [info]     [RtlSdr]      start stream
sdr-sdr-scanner-1         | Allocating 15 zero-copy buffers
sdr-sdr-scanner-1         | [2023-01-03 19:15:18.872] [auto_sdr] [info]     [Recorder]    create worker frequency: 444.000 kHz, total workers: 1, queue size: 0
sdr-sdr-scanner-1         | [2023-01-03 19:15:18.872] [auto_sdr] [info]     [RecorderWrk] start thread id: 24, frequency: 444.000 kHz
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.874][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:15:08.851 - 2023-01-03 19:15:17.290, 446000000 - 448000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.879][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (6, 8001), size: 48.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:18.940][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-scanner-1         | [2023-01-03 19:15:18.956] [auto_sdr] [info]     [DataCtrl]    start transmission frequency: 444.000 kHz
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.577][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:15:09.515 - 2023-01-03 19:15:19.253, 444000000 - 446000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.582][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (7, 8001), size: 56.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.594][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.898][mqtt_reader    ][Spectrogram  ][INFO   ] rtlsdr_00000001 save, 2023-01-03 19:15:09.849 - 2023-01-03 19:15:16.620, 448000000 - 450000000
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.902][mqtt_reader    ][Spectrogram  ][INFO   ] new data shape: (5, 8001), size: 40.0 kB
sdr-sdr-monitor-reader-1  | [2023-01-03 19:15:19.916][mqtt_reader    ][Spectrogram  ][INFO   ] save finish
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.663] [auto_sdr] [info]     [TornDtr]     report transmission frequency: 444.000 kHz, duration: 1.35 seconds
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.664] [auto_sdr] [info]     [DataCtrl]    finish transmission frequency: 444.000 kHz, duration: 1.28 seconds, reach minimum: false
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.664] [auto_sdr] [info]     [RecorderWrk] stop thread id: 24, frequency: 444.000 kHz, queue size: 0
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.664] [auto_sdr] [info]     [Recorder]    erase worker frequency: 444.000 kHz, total workers: 0, queue size: 0
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.725] [auto_sdr] [info]     [RtlSdr]      cancel stream
sdr-sdr-scanner-1         | [2023-01-03 19:15:20.764] [auto_sdr] [info]     [RtlSdr]      stop stream
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.023] [auto_sdr] [error]    [Scanner]     exception: read samples error
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.023] [auto_sdr] [info]     [Scanner]     stop thread id: 23
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.026] [auto_sdr] [info]     [Recorder]    stop thread id: 22
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.026] [auto_sdr] [info]     [SamplesWrk]  stop thread id: 18
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [spectrum]    deinit
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [SamplesWrk]  stop thread id: 19
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [spectrum]    deinit
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [SamplesWrk]  stop thread id: 20
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [spectrum]    deinit
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.027] [auto_sdr] [info]     [SamplesWrk]  stop thread id: 21
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.028] [auto_sdr] [info]     [spectrum]    deinit
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.045] [auto_sdr] [info]     [RtlSdr]      close device, index: 0, name: Generic RTL2832U OEM, serial: 00000001
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.129] [auto_sdr] [info]     [Mqtt]        stop thread id: 9
sdr-sdr-broker-1          | 1672769721: Client auto-98DF898A-08B3-4177-82ED-39B93D466E9C disconnected.
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.131] [auto_sdr] [info]     [main]        stop app auto_sdr
sdr-sdr-scanner-1         | [2023-01-03 19:15:21.131] [auto_sdr] [info]     [main]        stop thread id: 8
sdr-sdr-scanner-1 exited with code 0
`
And this is my config.json
`{
  "scanner_frequencies_ranges": [
    {
[... 51 lines cut ...]
  },
  "detection": {
    "frequency_grouping_size": 10000,
    "frequency_range_scanning_time_ms": 64,
    "noise_learning_time_seconds": 30,
    "noise_detection_margin": 10,
    "torn_transmission_learning_time_seconds": 60
  },
  "output": {
    "logs": "sdr/logs",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "cores": 4,
  "memory_limit_mb": 0,
  "mqtt": {
    "hostname": "sdr-broker",
    "port": 1883,
    "username": "admin",
    "password": "password"
  }
}
`

thank you very much

shajen, 2023-01-18:
Please update to the latest version and check if the error still occurs. If yes, please follow the instructions here and upload the full log file: https://github.com/shajen/rtl-sdr-scanner-cpp/tree/devel#debugging

jointerCZ, 2023-02-09:
I had the same or similar issue (but I did not saved the errors) when using multiple frequencies. The new build from yesterday seems to be fine so far. (VMWare Player, Ubuntu 22.04.1 and RTL-SDR from GIT) Thank you Shajen!
(I think there might be gain issue or changes with the new build, but it is too early to tell - it would be nice to have AGC/Tuner controls for the config)

shajen, 2025-10-20:
I added AGC/Tuner controls. Checkout the new version https://github.com/shajen/sdr-hub