shajen/rtl-sdr-scanner-cpp #15: procps2

geosphere, 2023-01-07:
Hi! Is it possible to add compatibility with the new procps (`procps2-dev`)?

shajen, 2023-01-18:
It actually works perfectly with `procps2`. What is your problem with `procps2`?

geosphere, 2023-01-19:
On Debian testing:

```
cmake -B build -DCMAKE_BUILD_TYPE=Release .
-- The CXX compiler identification is GNU 12.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Found RTLSDR: /usr/lib/x86_64-linux-gnu/librtlsdr.so
-- Found nlohmann_json: /usr/share/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.11.2") 
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.12.1")  
CMake Error at CMakeLists.txt:21 (message):
  procps library not found
```  
```sh
ls -la /lib/x86_64-linux-gnu/libproc*
-rw-r--r-- 1 root root 332686 Dec 19 08:06 /lib/x86_64-linux-gnu/libproc2.a
lrwxrwxrwx 1 root root     35 Dec 19 08:06 /lib/x86_64-linux-gnu/libproc2.so -> /lib/x86_64-linux-gnu/libproc2.so.0
lrwxrwxrwx 1 root root     17 Dec 19 08:06 /lib/x86_64-linux-gnu/libproc2.so.0 -> libproc2.so.0.0.1
-rw-r--r-- 1 root root 203272 Dec 19 08:06 /lib/x86_64-linux-gnu/libproc2.so.0.0.1
lrwxrwxrwx 1 root root     18 Oct 20 11:38 /lib/x86_64-linux-gnu/libprocps.so.8 -> libprocps.so.8.0.3
-rw-r--r-- 1 root root  84016 Oct 20 11:38 /lib/x86_64-linux-gnu/libprocps.so.8.0.3
```

```sh
ls -la /usr/include/libproc2 
total 112
drwxr-xr-x   2 root root  4096 Jan 19 03:44 .
drwxr-xr-x 151 root root 20480 Jan 19 03:44 ..
-rw-r--r--   1 root root  4632 Dec 19 08:06 diskstats.h
-rw-r--r--   1 root root 10098 Dec 19 08:06 meminfo.h
-rw-r--r--   1 root root  2889 Dec 19 08:06 misc.h
-rw-r--r--   1 root root 14082 Dec 19 08:06 pids.h
-rw-r--r--   1 root root  4921 Dec 19 08:06 slabinfo.h
-rw-r--r--   1 root root  6195 Dec 19 08:06 stat.h
-rw-r--r--   1 root root 24285 Dec 19 08:06 vmstat.h
-rw-r--r--   1 root root  6691 Dec 19 08:06 xtra-procps-debug.h
```

shajen, 2023-01-19:
This problem is related to your system because I built the application with docker on `debian:latest` instead of `ubuntu:22.04` and everything built without problems.

Try reinstalling `libprocps-dev` with `apt-get install --reinstall libprocps-dev`. The `pkgconfig` or `cmake` files of the `libprocps` package may be missing.

Check output of command `dpkg -L libprocps-dev`.

```
root@362507be15a1:/# dpkg -L libprocps-dev
/.
/usr
/usr/include
/usr/include/proc
/usr/include/proc/alloc.h
/usr/include/proc/devname.h
/usr/include/proc/escape.h
/usr/include/proc/numa.h
/usr/include/proc/procps.h
/usr/include/proc/pwcache.h
/usr/include/proc/readproc.h
/usr/include/proc/sig.h
/usr/include/proc/slab.h
/usr/include/proc/sysinfo.h
/usr/include/proc/version.h
/usr/include/proc/wchan.h
/usr/include/proc/whattime.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libprocps.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libprocps.pc
/usr/share
/usr/share/doc
/usr/share/doc/libprocps-dev
/usr/share/doc/libprocps-dev/NEWS.Debian.gz
/usr/share/doc/libprocps-dev/changelog.Debian.gz
/usr/share/doc/libprocps-dev/changelog.gz
/usr/share/doc/libprocps-dev/copyright
/usr/lib/x86_64-linux-gnu/libprocps.so
```

geosphere, 2023-01-20:
But docker with tag `debian:latest` is related to the `bullseye`, while in `bookworm`  the library `libprocps` (package  `libprocps-dev`) is obsolete and was replaced by `libproc2` (package `libproc2-dev`)

```
dpkg -L libproc2-dev  
/.
/usr
/usr/include
/usr/include/libproc2
/usr/include/libproc2/diskstats.h
/usr/include/libproc2/meminfo.h
/usr/include/libproc2/misc.h
/usr/include/libproc2/pids.h
/usr/include/libproc2/slabinfo.h
/usr/include/libproc2/stat.h
/usr/include/libproc2/vmstat.h
/usr/include/libproc2/xtra-procps-debug.h
/usr/lib
/usr/lib/valgrind
/usr/lib/valgrind/libproc.supp
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libproc2.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libproc2.pc
/usr/share
/usr/share/doc
/usr/share/doc/libproc2-dev
/usr/share/doc/libproc2-dev/NEWS.Debian.gz
/usr/share/doc/libproc2-dev/changelog.Debian.gz
/usr/share/doc/libproc2-dev/changelog.gz
/usr/share/doc/libproc2-dev/copyright
/usr/lib/x86_64-linux-gnu/libproc2.so
```

shajen, 2023-01-23:
I removed `libprocsps` dependency. For now is available only on `devel` branch, so you have to use `devel` branch or cherry-pick commit `0d54dbe7def7bd66c838d8be03cc65eee3fa809b` or wait for merge `devel` to master.

shajen, 2023-02-07:
It's was fixed and merged to master. Please update to latest version.

geosphere, 2023-02-08:
Many thanks! I confirm that now the compilation passes without errors

closed 2023-02-08 by geosphere