shajen/rtl-sdr-scanner-cpp #16: Using step of 2500 not working

smokedsalmonbagel, 2023-01-18:
Firstly, this a really great application - thanks!

I had tried to use a step size of 2500.  Took me a while to figure out what was causing no `rtl-scanner-sdr-monitor-reader` output. Seemed like no scanning was happening. When I changed it back to 1000 it works fine.  Am I misunderstanding what step means here?
Did not work:
```
{
      "device_serial": "02",
      "ranges": [
        {
          "start": 115000000,
          "stop": 136000000,
          "step": 2500,
          "sample_rate": 2048000
        }
      ]
    },
```
Worked:
```
{
      "device_serial": "02",
      "ranges": [
        {
          "start": 115000000,
          "stop": 136000000,
          "step": 1000,
          "sample_rate": 2048000
        }
      ]
    },
```

Can post my full config and console output but nothing seemed to stand out.  Just no output from the `rtl-scanner-sdr-monitor-reader` container.

shajen, 2023-01-18:
See the manual here, it explains corelaction between `step` and `sample_rate`:
https://github.com/shajen/rtl-sdr-scanner-cpp#custom-sample-rate-step-and-frequency-range

smokedsalmonbagel, 2023-01-18:
Ok , yes I saw that but was not sure what `n` is here:

`Please note that sample_rate must fit to step. You should meet the following equation sample_rate / step = 2 ^ n.`

shajen, 2023-01-18:
`n` is any value that satisfies the equation.

smokedsalmonbagel, 2023-01-18:
I see the issue, I miscounted the number of zeros.  Thanks for the quick reply!

closed 2023-01-18 by smokedsalmonbagel

wbwarnerb, 2023-02-11:
Hi - could either of you help me with regards to step/sampling?  I'd like to pick up a freq like 470.5375 is this doable?  I notice that with the default setup of 2048000 sample rate and a step 1000, this frequency is not caught.  I'm not sure what values would be required to narrow to *.5375.