shajen/rtl-sdr-scanner-cpp #17: Supported modes

smokedsalmonbagel, 2023-01-19:
What are the supported modes?  I have been scanning in the air band but haven't found any signals.  Is AM supported?  Have there been any tests with the AM mode?

shajen, 2023-01-23:
AM is not supported. Transmissions on aircraft airband is marked as AM but there is no decoder for it yet. For now, only FM is supported.

TechMindsYT, 2023-02-02:
Will AM be supported soon? This would be a good feature for aircraft enthusiasts. Thanks

jacko0, 2023-02-03:
I would also like to listen to AM transmissions.

smokedsalmonbagel, 2023-02-03:
> AM is not supported. Transmissions on aircraft airband is marked as AM but there is no decoder for it yet. For now, only FM is supported.

Understood.  There is probably a decent number of people who would like this feature to find military aircraft transmissions etc.

shajen, 2023-02-07:
Yeah I know. I'm working on AM demodulation. It will be added in the future.

shajen, 2023-02-15:
I have good news. I added support for AM modulation in the latest version.

Install and update instruction [here](https://github.com/shajen/rtl-sdr-scanner-cpp/wiki/Quickstart).

closed 2023-10-01 by shajen