shajen/rtl-sdr-scanner-cpp #18: Timezone

pinozgit, 2023-02-04:
sdr-sdr-monitor-reader-1  | ValueError: Incorrect timezone setting: -03

wbwarnerb, 2023-02-04:
I'm also getting this on the docker-compose install.  cat /etc/timezone is America/Los_Angeles and during docker-compose I get a stack trace and ultimately: 

sdr-monitor-server_1  |     raise ValueError("Incorrect timezone setting: %s" % self.TIME_ZONE)
sdr-monitor-server_1  | ValueError: Incorrect timezone setting: PST

rodgerhanson, 2023-02-07:
I'm getting this error about timezone setting when doing the "docker compose up" command:
ValueError: Incorrect timezone setting: NZDT

How can I fix this one please?

[image]

shajen, 2023-02-07:
It's was fixed. Please update to latest version. Remember to update docker images, `config.json` and `docker-compose.yml`.

If any problem with timezone settings, see https://github.com/shajen/rtl-sdr-scanner-cpp#timezone

closed 2023-02-15 by shajen