shajen/rtl-sdr-scanner-cpp #20: Build and start works fine on Ubuntu 20 but not sure how to get to Admin console

antonjan, 2023-02-05:
Running the start up command in Ubuntu has a spelling mistake 
it si ./build/auto-sdr config.json   in docs but should be as below "-" hyphen should be underscore "_"
./build/auto_sdr config.json

All start up and it scans the frequency but how do I get to the Admin console.
The program dose not seem to listen on 8088 or 8000 where can I get or set the port details or do I need to run something else.
I notes that the program is doing something with MQTT but not sure what to do next to get the admin console?

jointerCZ, 2023-02-06:
You probably compiled it directly as I did, then I had the same experience. It is probably missing some subcomponents like Mosquitto, I used the Docker guide and that works fine, based on 1 day experience. Running in VMWare Player.

shajen, 2023-02-07:
@jointerCZ
You are right. Docker version has all in one. 

@antonjan
If you want run `sdr-scanner` from sources you have to run only `mqtt-broker` and `sdr-monitor` in docker and connect `sdr-scanner` to them (you have to change `mqtt` settings in `config.json`).

pdxh0b0, 2023-02-08:
> Running the start up command in Ubuntu has a spelling mistake it si ./build/auto-sdr config.json in docs but should be as below "-" hyphen should be underscore "_" ./build/auto_sdr config.json
>
> All [...]

Something I haven't used in quite some time was messing me about like you've described,  icecast server running in the background(using 8000)
sudo systemctl stop icecast2
Try running the scanner again

closed 2023-11-13 by shajen