shajen/rtl-sdr-scanner-cpp #22: Rpi

blackwellj, 2023-02-05:
Has anyone had any luck running this on a Pi? I cannot get it to work unfortunatly

folkertvanheusden, 2023-02-07:
Could you please be more specific?

blackwellj, 2023-02-07:
It only seems to work on an amd processor. The pi uses a arm processor

shajen, 2023-02-07:
It's work perfectly on rpi with sample rate `10 Mhz`. You have to build scanner from sources and connect to mqtt broker and web interface run on another machine.

skandragon, 2023-02-08:
Any particular reason why mqtt and the UI can't also run on the Pi, other than the sdr images aren't multi-architecture?

shajen, 2023-02-09:
I released docker version of all components (`sdr-scanner`, `sdr-monitor`, `mqtt-broker`) for `arm32`, `arm64`, `amd64`. Now it should work on `RaspberryPi`. Strongly recommend to familiar with [RaspberryPi](https://github.com/shajen/rtl-sdr-scanner-cpp#raspberrypi) section before start.

Checkout latest version.

blackwellj, 2023-02-10:
sdr-monitor-reader The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested  0.0s

Im getting this error on the pi when i try to run it

shajen, 2023-02-10:
In #30 you can check that it works on `RaspberryPi`. I think you have old image. Remove image and download new or update image by `docker compose pull`.

skandragon, 2023-02-13:
The images seem to be there, but I've not yet tried them.  I will next weekend if not sooner.   Thanks!

closed 2023-10-01 by shajen