shajen/rtl-sdr-scanner-cpp #29: UI not being served

oldirty91, 2023-02-09:
Hello and thank you for what seems like a great piece of sw. Having issues getting to the localhost for the UI. I followed the steps to build locally with docker. After messing with the typical rtl-sdr blacklist settings I was able to get it running using `docker compose up`. It is clearing running in the command line and I can even see it recording transmissions when keying up my radio, however, I can not get to the UI on the localhost. I tried 0.0.0.0:8000, 127.0.0.1:8000, localhost:8000, and many more with no joy.

Are there additional steps required when building locally that are required to get the web server running? logs.txt is attached below.

https://file.io/fdes0mYtFXx0

DereIBims, 2023-02-09:
I have the same problem on an Pi 4.
Seems like the web host (sdr-monitor) is not running on ARM.

shajen, 2023-02-09:
I released docker version of all components (`sdr-scanner`, `sdr-monitor`, `mqtt-broker`) for `arm32`, `arm64`, `amd64`. Now it should work on `RaspberryPi`. Strongly recommend to familiar with [RaspberryPi](https://github.com/shajen/rtl-sdr-scanner-cpp#raspberrypi) section before start.

Checkout latest version.

closed 2023-11-13 by shajen