shajen/rtl-sdr-scanner-cpp #30: Not Running on Raspberry Pi 4

DereIBims, 2023-02-09:
Hello,

docker-compose successfully downloads and creates all 3 containers.
But sdr_scanner and sdr_monitor keep restarting with the same error " ".

Builing sdr_scanner myself fixed the problem for that container, but there's no way to also simply build sdr_monitor.

Please provide docker images for ARM aswell or a guide to build myself.

Thank you!

shajen, 2023-02-09:
I released docker version of all components (`sdr-scanner`, `sdr-monitor`, `mqtt-broker`) for `arm32`, `arm64`, `amd64`. Now it should work on `RaspberryPi`. Strongly recommend to familiar with [RaspberryPi](https://github.com/shajen/rtl-sdr-scanner-cpp#raspberrypi) section before start.

Checkout latest version.

DereIBims, 2023-02-09:
Works now without any problems on an PI4 4Gb.
Thank you very much!

closed 2023-02-10 by DereIBims