shajen/rtl-sdr-scanner-cpp #31: Error: usb_claim_interface error -6

wbwarnerb, 2023-02-09:
I hope you can help me with this one.  I've now successfully installed the software on Ubuntu and on Debian (two different machines) with the latest pull and I get the same problem with my RTL-SDRs (I have two that have been tested). 

After a few minutes of running the app, it stops collecting transmissions, and when I check the docker logs, I see usb_claim_interface error -6 followed by an exit code 0.  If I try and restart or stop/start docker, it refuses to start after that, citing the same error in the logs, exiting out.  

I'm able to use other software with the RTL-SDR (such as dump1090), to test that it isn't the issue.  

Once this error state occurs, the only way to mitigate it appears to be rebooting the system.  After a full reboot the interface claim is removed and I can run the software again, but it will soon hit this error after several (about 7-8) audio files are processed.  I was scanning 470-472Mhz which had quite activity (about 7 broadcasts a min).

shajen, 2023-02-09:
Please reproduce problem and provide debug logs basing on [instruction](https://github.com/shajen/rtl-sdr-scanner-cpp#debugging).

wbwarnerb, 2023-02-09:
Thanks for the response.  After I put it in trace, oddly it has been working.  both machines are working without issue right now.  I'm going to close this since I can't reproduce it right now, and will monitor to see if it's something related to output. I'm running it on a server, which should be fine. If I can reproduce I'll open a new task.  Thanks for your help.

closed 2023-02-09 by wbwarnerb