shajen/rtl-sdr-scanner-cpp #33: Error: rtlsdr_demod_read_reg failed with -7

jointerCZ, 2023-02-09:
Hello.
I upgraded to the latest version and when I have multiple band scanned with one SDR, it no longer crashes after the first recording.
But it still crashes for me, it lasted about 5 hours this time:

sdr-sdr-monitor-reader-1  | [2023-02-09 23:40:53.164][mqtt_reader    ][Spectrogram  ][INFO   ] save, rtlsdr_00000001, 2023-02-09 23:40:53.109, 438000000 - 440000000, 2.0 kB, -65.14 dBfs
sdr-sdr-monitor-reader-1  | [2023-02-09 23:40:53.466][mqtt_reader    ][Spectrogram  ][INFO   ] save, rtlsdr_00000001, 2023-02-09 23:40:53.399, 445000000 - 447000000, 2.0 kB, -59.49 dBfs
sdr-sdr-scanner-1         | rtlsdr_demod_read_reg failed with -7
sdr-sdr-scanner-1         | r82xx_write_arr: i2c wr failed=-9 reg=17 len=1
sdr-sdr-scanner-1         | rtlsdr_demod_write_reg failed with -9
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.749325] [error  ] [Scanner    ] exception: set center frequency error
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.749352] [info   ] [Scanner    ] stop thread id: 23
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759626] [info   ] [SamplesWrk ] stop thread id: 19
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759736] [info   ] [spectrogram] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759750] [info   ] [fft        ] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759882] [info   ] [SamplesWrk ] stop thread id: 20
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759980] [info   ] [spectrogram] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.759993] [info   ] [fft        ] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760074] [info   ] [SamplesWrk ] stop thread id: 21
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760154] [info   ] [spectrogram] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760164] [info   ] [fft        ] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760249] [info   ] [SamplesWrk ] stop thread id: 22
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760352] [info   ] [spectrogram] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.760410] [info   ] [fft        ] deinit
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.771296] [info   ] [RtlSdr     ] close device, index: 0, name: Generic RTL2832U OEM, serial: 00000001
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.929286] [info   ] [Mqtt       ] stop thread id: 9
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.929655] [info   ] [main       ] stop app auto_sdr
sdr-sdr-scanner-1         | [2023-02-09 23:40:55.929674] [info   ] [main       ] stop thread id: 7
sdr-sdr-broker-1          | 1675982455: Client auto-194E1C2E-EA88-AEEE-C231-86CA0090E3B1 disconnected.
sdr-sdr-scanner-1 exited with code 0

^CGracefully stopping... (press Ctrl+C again to force)
Aborting on container exit...

I set the log level to trace and will provide full log once it happens again.
Device is normal cheap dongle with T820, Ubuntu 22.04.1 in VMWare Player (VM have 8 of AMD 5950X cores + 16GB RAM) with RTL-SDR about 5 days old from GIT. I will later configure it on old laptop to see if the VM is the problem.

Otherwise I like the SW very much, but I would appreciate a full list of possible config values to be documented.
(and of course other demodulators later)

shajen, 2023-02-10:
Very interesting. Please run docker in background mode with `docker compose up -d`. If `sdr-scanner` crash, docker should restart app and log this message to logs. Please send log after a few days and I can check how often it happens. Is it very possible that this is random bug with `USB bus` or something similar. You don't need to worry about it because docker should restart app and continue work.

Use this instruction to get log https://github.com/shajen/rtl-sdr-scanner-cpp#debugging `trace` level is not needed, `info` level is enough

jointerCZ, 2023-02-12:
Right now it is still running since Friday (1 SDR unit switching between 3 bands (2m/70cm/PMR) with trace and not with -d and did not crashed yet. If it crashes, I will send log and then rerun with only info debug level. But I did seen this issue at least twice.

BTW the spectrogram image layout is shifted (not aligned with the frequency image on top) when I am running 400kHz wide band only for PMR.

closed 2023-02-15 by jointerCZ