shajen/rtl-sdr-scanner-cpp #34: ignore frequencies failing

pdxh0b0, 2023-02-10:
log of failed run with ignored frequencies
https://justpaste.it/aribq

config with ignored frequencies added
https://justpaste.it/2x752

working config with two sdr's
https://justpaste.it/anth4

appreciate your help/insight on where I am going wrong in my edits

shajen, 2023-02-10:
Your config with ignored frequencies is invalid, it's not correct `json` file and can not be parsed by `sdr-scanner`. Paste here http://json.parser.online.fr/ and check.

pdxh0b0, 2023-02-10:
> Your config with ignored frequencies is invalid, it's not correct `json` file and can not be parsed by `sdr-scanner`. Paste here http://json.parser.online.fr/ and check.

Thank you, up and running 😻

closed 2023-02-15 by shajen