shajen/rtl-sdr-scanner-cpp #35: Single docker container

blackwellj, 2023-02-10:
Would it be possible to have a single docker container? I really don’t like having so many different containers. I want to run it on unraid which isn’t possible currently

shajen, 2023-02-10:
I don't think is it good idea. The idea of docker is to build multiple independent small container where everyone has single responsibility and connect them. One big container that have all in one is the opposite of the idea of docker.

Moreover, the idea of this project is to run multiple `sdr-scanner`s in many different physical locations and one `sdr-monitor` and one `mqtt-broker` in the cloud and connect all `sdr-scanner` to them.

With one big container is not possible.

blackwellj, 2023-02-11:
It might be a good idea to specify the imagined use case, I didnt realise that was the use for this software

algorni, 2023-02-13:
+1 on explaining a bit more the use case ;)

Am i assuming correctly that the srd-scanner componend didn't store anything? it just rely on the broker to stream the data to the sdr-monitor wich will be the solo responsible to to store the recording?

shajen, 2025-10-20:
I decide to simplify the project. Checkout the new version https://github.com/shajen/sdr-hub