shajen/rtl-sdr-scanner-cpp #41: Tips for packet transmissions?

lonseidman, 2023-02-11:
First a big thank you to Shajen for an awesome app. I have been playing with this all week and had some great results on the 2 meter band listening for the most active repeaters in my area.

What I'd like to do next is have it scan popular packet radio frequencies (including APRS) to see what kind of activity is happening around me. So far I've been able to get only one packet transmission recorded (my own) but haven't been able to get anything else to save. I reduced the minimum record time to see if maybe the packet transmissions are too short to get picked up.

Any thoughts or tips?

shajen, 2023-02-12:
I think that you have to set custom gain in config because default gain will detect only really strong signal.

shajen, 2023-02-12:
I'm not familiar with APRS. I think that repeaters usually has better localization, better antenna and are much stronger than individual APRS station. So receive of repeater is much easier that APRS. 

Maybe you need set correct gain? Maybe better antena? Or in your area there is no any APRS fan? Hard to say.