shajen/rtl-sdr-scanner-cpp #46: Armbian ARM64 attempt to install error

zero48, 2023-02-14:
Here is my attempt to install the system in ARMBIAN 64
i get this error after all containers are pulled (except this one)
Attaching to sdr-sdr-broker-1, sdr-sdr-monitor-reader-1, sdr-sdr-monitor-server-1, sdr-sdr-monitor-setup-1, sdr-sdr-scanner-1
Error response from daemon: failed to create endpoint sdr-sdr-broker-1 on network sdr_default: failed to add the host (veth170fb26) <=> sandbox (veth14dac1a) pair interfaces: operation not supported

Any hint ?

Thanks in advance

zero48, 2023-02-15:
I did a reboot and it is working as expected... By the way it is a 4/64gb tvbox flashed in EMMC with armbian ... I'll leave it running and close this issue after 24H ... if everything goes accordingly

zero48, 2023-02-15:
Transmissions count: 1228 in 24 h 3 different frequencies ... Not bad for a $35 server ... Wondering where is the path for the saved mp3 ?

shajen, 2023-02-15:
mp3 ale available in panel https://github.com/shajen/rtl-sdr-scanner-cpp#panel

zero48, 2023-02-15:
I meant in the containner ...

shajen, 2023-02-15:
`/usr/src/sdr-panel/public/media`

zero48, 2023-02-15:
U rock Thanks  !! I'll close this since is working correctly in Armbian64 .

closed 2023-02-15 by zero48