shajen/rtl-sdr-scanner-cpp #47: Transmission deletion fails over 1000 Files

Klammraute, 2023-02-15:
Hi,
I tried to delete all ca. 15000 Transmissions files in my Database but if i select all 15000 Files at once and try to delete i get a "400 bad request"

So i delete 14000 in 100 Steps and the last 1000 i could delete all at once. 

Thank you for your work!

shajen, 2023-02-16:
Please attach logs. Details [here](https://github.com/shajen/rtl-sdr-scanner-cpp#debugging).

jointerCZ, 2023-02-16:
I get the same error with about 4000 items.
This is error from the log:
sdr-sdr-monitor-server-1  | [2023-02-16 13:25:16.758][exception      ][django.security.TooManyFieldsSent][ERROR  ] The number of GET/POST parameters exceeded settings.DATA_UPLOAD_MAX_NUMBER_FIELDS.

shajen, 2023-02-17:
It looks like you have really much data to delete which exceed the http limit.

Maybe try do delete `device` in https://localhost:8000/admin/sdr/device/ ? It will cause cascade delete off all recordings and spectrograms belonging to this device.

jointerCZ, 2023-02-17:
Is this some internal limit of the Django or can be configured?
Lot of short transmissions are recorded, as I still work on defining ignored frequencies, I am at 2000 after 1 day or so.
Maybe "mark oldest 900" button can be added to work around this? Instead of mark all.
Will try the device delete next time.

jointerCZ, 2023-02-21:
Device delete is a workaround. The same device will then get registered under new ID.

tubalainen, 2023-03-02:
Same here, tried to delete 100k+ transmissions and got "Bad Request (400)" as an error.
[image]