shajen/rtl-sdr-scanner-cpp #48: Feature requests / Suggestions

jointerCZ, 2023-02-15:
Hi Shajen,
first of all, let me thank you for great piece of software, I will probably donate to support you.
I do not know how much time or passion you have for it, but let me suggest some UI improvements for the future.

1. it would be great if you could implement frequency mapping to a name. So in the Frequency column, we could see a name for a frequency. In some external file which can be given to the container on load so it can be easily transferred maybe? Or in the Admin section, but with possibility to import and export.
2. the slider for moving time in the recording should be much wider, to be able better move inside the recording. But it seems this is some kind of browser feature, because the column "Decoded file" have a different appearance in Edge and in Firefox. So not sure if that would be possible.
3.  it would be great to be able to see ignored frequencies in the spectrogram. Something like a grey line overlay over the whole spectrogram. Or if that would be difficult to do, a mark on the top gray frequency bar (like a red dot or bar or triangle) would be also great. Would be helpful when analyzing the spectrograms.
4. the AM decoder would be nice for people who want to record the air band.
5. maybe some kind of GUI configuration tool for the config file with all the possible options, but that is the last thing to do (after everything else is done), because people will probably want it for multiple platforms including Windows, for cases where is no X server on their Linux boxes.

Nothing else comes to mind after few days of use, which I think is a good thing.
73

shajen, 2023-02-16:
Thank you for kind words :)

> 1. it would be great if you could implement frequency mapping to a name. So in the Frequency column, we could see a name for a frequency. In some external file which can be given to the container on [...]

Great feature. I have similar thoughts. I plan to add a frequency name for each range in `config.json` and show it in new column in panel.
>     2. the slider for moving time in the recording should be much wider, to be able better move inside the recording. But it seems this is some kind of browser feature, because the column "Decoded file" have a different appearance in Edge and in Firefox. So not sure if that would be possible.

It will be a bit difficult because this element is rendered by browser and can not be customized. I will see what can I do.
> 3. it would be great to be able to see ignored frequencies in the spectrogram. Something like a grey line overlay over the whole spectrogram. Or if that would be difficult to do, a mark on the top [...]

If it's easy and more people want it, I'll add it.
>     4. the AM decoder would be nice for people who want to record the air band.

I have already added the AM decoder.
>     5. maybe some kind of GUI configuration tool for the config file with all the possible options, but that is the last thing to do (after everything else is done), because people will probably want it for multiple platforms including Windows, for cases where is no X server on their Linux boxes.

A very useful tool, but certainly the most work-intensive.

jointerCZ, 2023-02-16:
OK, another question.
I have currently 2 RTL SDR connected, is it possible to configure PPM and Gain for them independently?
Each is for different band.
I want to eventually connect 3rd but that would need a PPM correction.

HarbingerSe7en, 2023-02-16:
Is there any planned support for additional SDRs? Namely Airspy and BladeRF.

shajen, 2023-02-17:
> OK, another question. I have currently 2 RTL SDR connected, is it possible to configure PPM and Gain for them independently? Each is for different band. I want to eventually connect 3rd but that would need a PPM correction.

Unfortunately no. This is a bug of architecture/config. I'll add to the to-do list.

> Is there any planned support for additional SDRs? Namely Airspy and BladeRF.

Yes, but rather in the distant future.

jointerCZ, 2023-02-18:
Have another, by exploring the digital noises:
DMR / D-Star / C4FM decoders...

shajen, 2023-02-19:
Yeah I know. But decoding digital transmissions will take a lot of work. If I found libraries that are able to decode digital signals I will use them.

HarbingerSe7en, 2023-02-19:
> Yeah I know. But decoding digital transmissions will take a lot of work. If I found libraries that are able to decode digital signals I will use them.

Would you be able to integrate [DSD-FME](https://github.com/lwvmobile/dsd-fme) decoders into the project?