shajen/rtl-sdr-scanner-cpp #49: Question: The path to recorded audio files?

atlex, 2023-02-18:
Hi @shajen,

How can i configure the path to recorded audio files?

Old setting from config.json doesn't work.
 
"output": {
    "recordings": "/home/pi/recs",

tomevans80, 2023-02-19:
Not sure how to configure it but mine are in /var/snap/docker/common/var-lib-docker/volumes/

shajen, 2023-02-19:
You can not configure path of recordings. The are 3 ways to access recordings:

- http://localhost:8000/sdr/transmissions
- inspect docker volume `sdr_monitor_data`
- directory `/usr/src/sdr-panel/public/media` in `sdr_sdr-monitor-reader_1` container

atlex, 2023-02-20:
@shajen thank you.

I found recorded files (.bin) in the sdr_sdr-monitor-reader_1 container.
But it's not useful for me :(

Is it possible to have recorded audio files (.mp3/.wav) available for other applications on the host machine?

atlex, 2023-02-22:
@shajen Now i have recorded files in `/home/den/rtl-sdr-scanner-cpp/data/device_2/transmission/2023-02-22/`

My solution:

docker-compose.yml

```
services:
  sdr-broker:
    image: eclipse-mosquitto:latest
    restart: on-failure
    command:
      - /bin/sh
      - -c
      - |
        echo "listener 1883" > /mosquitto/config/mosquitto.conf
        echo "allow_anonymous false" >> /mosquitto/config/mosquitto.conf
        echo "password_file /mosquitto/config/password.txt" >> /mosquitto/config/mosquitto.conf
        echo "admin:password" > /mosquitto/config/password.txt
        mosquitto_passwd -U /mosquitto/config/password.txt
        mosquitto -c /mosquitto/config/mosquitto.conf
  sdr-scanner:
    image: shajen/sdr-scanner:latest
    restart: on-failure
    depends_on:
      - sdr-broker
    volumes:
      - /etc/timezone:/etc/timezone:ro
      - /etc/localtime:/etc/localtime:ro
      - ./config.json:/config.json
    devices:
      - /dev/bus/usb:/dev/bus/usb
  sdr-monitor-setup:
    image: shajen/sdr-monitor:latest
    restart: on-failure
    depends_on:
      - sdr-broker
    volumes:
      - /etc/timezone:/etc/timezone:ro
      - /etc/localtime:/etc/localtime:ro
      - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
      - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
    command: bash -c "python3 manage.py migrate && DJANGO_SUPERUSER_PASSWORD=password python3 manage.py createsuperuser --noinput --username admin --email admin@local.local &> /dev/null || true"
  sdr-monitor-server:
    image: shajen/sdr-monitor:latest
    restart: on-failure
    depends_on:
      sdr-monitor-setup:
        condition: service_completed_successfully
    volumes:
      - /etc/timezone:/etc/timezone:ro
      - /etc/localtime:/etc/localtime:ro
      - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
      - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
    command: python3 manage.py runserver 0.0.0.0:8000
    ports:
      - 8001:8000
  sdr-monitor-reader:
    image: shajen/sdr-monitor:latest
    restart: on-failure
    depends_on:
      sdr-monitor-setup:
        condition: service_completed_successfully
    volumes:
      - /etc/timezone:/etc/timezone:ro
      - /etc/localtime:/etc/localtime:ro
      - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
      - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
    command: python3 manage.py runscript mqtt_reader
```

algorni, 2023-02-27:
where is the source code of manage.py ?

abdellahaski, 2023-09-28:
> @shajen Now i have recorded files in my local folder `/home/den/rtl-sdr-scanner-cpp/data/device_2/transmission/2023-02-22/`
> 
> My solution:
> 
> docker-compose.yml
> 
> ```
> services:
>   sdr-broker:
>     image: eclipse-mosquitto:latest
>     restart: on-failure
>     command:
>       - /bin/sh
>       - -c
>       - |
>         echo "listener 1883" > /mosquitto/config/mosquitto.conf
>         echo "allow_anonymous false" >> /mosquitto/config/mosquitto.conf
>         echo "password_file /mosquitto/config/password.txt" >> /mosquitto/config/mosquitto.conf
>         echo "admin:password" > /mosquitto/config/password.txt
>         mosquitto_passwd -U /mosquitto/config/password.txt
>         mosquitto -c /mosquitto/config/mosquitto.conf
>   sdr-scanner:
>     image: shajen/sdr-scanner:latest
>     restart: on-failure
>     depends_on:
>       - sdr-broker
>     volumes:
>       - /etc/timezone:/etc/timezone:ro
>       - /etc/localtime:/etc/localtime:ro
>       - ./config.json:/config.json
>     devices:
>       - /dev/bus/usb:/dev/bus/usb
>   sdr-monitor-setup:
>     image: shajen/sdr-monitor:latest
>     restart: on-failure
>     depends_on:
>       - sdr-broker
>     volumes:
>       - /etc/timezone:/etc/timezone:ro
>       - /etc/localtime:/etc/localtime:ro
>       - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
>       - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
>     command: bash -c "python3 manage.py migrate && DJANGO_SUPERUSER_PASSWORD=password python3 manage.py createsuperuser --noinput --username admin --email admin@local.local &> /dev/null || true"
>   sdr-monitor-server:
>     image: shajen/sdr-monitor:latest
>     restart: on-failure
>     depends_on:
>       sdr-monitor-setup:
>         condition: service_completed_successfully
>     volumes:
>       - /etc/timezone:/etc/timezone:ro
>       - /etc/localtime:/etc/localtime:ro
>       - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
>       - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
>     command: python3 manage.py runserver 0.0.0.0:8000
>     ports:
>       - 8001:8000
>   sdr-monitor-reader:
>     image: shajen/sdr-monitor:latest
>     restart: on-failure
>     depends_on:
>       sdr-monitor-setup:
>         condition: service_completed_successfully
>     volumes:
>       - /etc/timezone:/etc/timezone:ro
>       - /etc/localtime:/etc/localtime:ro
>       - /home/den/rtl-sdr-scanner-cpp/db:/usr/src/sdr-panel/db
>       - /home/den/rtl-sdr-scanner-cpp/data:/usr/src/sdr-panel/public/media
>     command: python3 manage.py runscript mqtt_reader
> ```

But how do you read the bin files

GarryCZ, 2023-11-06:
If anyone needs to pull MP3s from the server, I've written a primitive script that downloads the last file and saves it to the sounds folder when a new recording is made. I'm sure there are other ways to do this, but this is sufficient for me and works well. 

code : https://pastebin.com/raw/sD67WXcp

closed 2023-11-13 by shajen