shajen/rtl-sdr-scanner-cpp #50: Using SoapySDR as backend

MrTalon63, 2023-02-18:
First of all, great product I was playing with it since yesterday and I love it. But I was wondering if it was possible to use SoapySDR as the backend thus enabling pretty much every SDR device to work. I stumbled across another issue/feature request that asked for RSP support. I'm aware that integrating every single piece of hardware is just not viable for a small personal project. I'm not fully aware of how the project is structured and if it's even possible to implement it but I would guess it would be a game changer.

shajen, 2023-02-19:
Great idea, thanks for the info. It should be doable. I will try to use this library in the near future. I wish I had found this library sooner.

shajen, 2023-11-13:
Check new devel version: https://github.com/shajen/rtl-sdr-scanner-cpp/issues/76#issuecomment-1787475909

closed 2023-11-13 by shajen