shajen/rtl-sdr-scanner-cpp #51: sdr-scanner crashes "cannot start stream"

skandragon, 2023-02-18:
I am having some sort of problem, either configuration or code, that causes sdr-scanner to crash after a while.  Spectrum is graphed (although it looks... strange) but no transmissions have been received.  I suspect, but have not yet verified, that receiving a transmission causes sdr-scanner to crash.

I'm using a HackRF with the latest firmware:

```
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82876671b
Board ID Number: 2 (HackRF One)
Firmware Version: 2023.01.1 (API:1.07)
Part ID Number: 0xa000cb3c 0x00704358
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```

## Config:

```
{
  "scanner_frequencies_ranges": [
    {
      "device_serial": "auto",
      "ranges": [
        {
          "start": 144000000,
          "stop": 146000000,
          "sample_rate": 2048000
        },
        {
          "start": 128000000,
          "stop": 130000000,
          "sample_rate": 2048000
        }
      ]
    }
  ],
  "ignored_frequencies": [],
  "devices": {
    "rtl_sdr": {
      "ppm_error": 0,
      "tuner_gain": 0.0,
      "offset": 0
    },
    "hack_rf": {
      "lna_gain": 16,
      "vga_gain": 42,
      "offset": 0
    }
  },
  "recording": {
    "max_noise_time_ms": 500,
    "min_time_ms": 2000,
    "min_sample_rate": 16000
  },
  "detection": {
    "frequency_grouping_size": 10000,
    "frequency_range_scanning_time_ms": 64,
    "noise_learning_time_seconds": 30,
    "noise_detection_margin": 10,
    "torn_transmission_learning_time_seconds": 60
  },
  "output": {
    "logs": "sdr/logs",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "cores": 4,
  "memory_limit_mb": 0,
  "mqtt": {
    "hostname": "sdr-broker",
    "port": 1883,
    "username": "admin",
    "password": "password"
  }
}

```

## Logs

```
sdr-scanner_1         | [2023-02-18 15:29:14.465032] [info   ] [HackRf     ] start stream, samples: 262144
sdr-scanner_1         | [2023-02-18 15:29:14.547823] [info   ] [HackRf     ] stop stream
sdr-scanner_1         | [2023-02-18 15:29:14.560697] [info   ] [HackRf     ] start stream, samples: 262144
sdr-monitor-reader_1  | [2023-02-18 15:29:14.595][mqtt_reader    ][Spectrogram  ][INFO   ] save, hackrf_a06063c82876671b, 2023-02-18 15:29:14.559, 144000000 - 146000000, 2.0 kB, -72.59 dBfs
sdr-scanner_1         | [2023-02-18 15:29:14.644670] [info   ] [HackRf     ] stop stream
sdr-scanner_1         | [2023-02-18 15:29:14.662337] [info   ] [HackRf     ] start stream, samples: 262144
sdr-scanner_1         | [2023-02-18 15:29:14.663258] [error  ] [Scanner    ] exception: can not start stream
sdr-scanner_1         | [2023-02-18 15:29:14.663370] [info   ] [Scanner    ] stop thread id: 18
sdr-scanner_1         | [2023-02-18 15:29:14.672394] [info   ] [SamplesWrk ] stop thread id: 14
sdr-scanner_1         | [2023-02-18 15:29:14.672656] [info   ] [spectrogram] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.672748] [info   ] [fft        ] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.672834] [info   ] [SamplesWrk ] stop thread id: 15
sdr-scanner_1         | [2023-02-18 15:29:14.672944] [info   ] [spectrogram] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.673026] [info   ] [fft        ] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.673115] [info   ] [SamplesWrk ] stop thread id: 16
sdr-scanner_1         | [2023-02-18 15:29:14.673238] [info   ] [spectrogram] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.673323] [info   ] [fft        ] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.673415] [info   ] [SamplesWrk ] stop thread id: 17
sdr-scanner_1         | [2023-02-18 15:29:14.673717] [info   ] [spectrogram] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.673839] [info   ] [fft        ] deinit
sdr-scanner_1         | [2023-02-18 15:29:14.674132] [info   ] [HackRf     ] close device, serial: a06063c82876671b
sdr-scanner_1         | [2023-02-18 15:29:15.188436] [info   ] [HackRf     ] exit
sdr-scanner_1         | [2023-02-18 15:29:15.294856] [info   ] [Mqtt       ] stop thread id: 10
sdr-scanner_1         | [2023-02-18 15:29:15.295698] [info   ] [main       ] stop app auto_sdr
sdr-scanner_1         | [2023-02-18 15:29:15.295766] [info   ] [main       ] stop thread id: 7
sdr-broker_1          | 1676755755: Client auto-3C730CAA-260C-3D3C-8FA6-F78C0B869E55 disconnected.
sdr-scanner_sdr-scanner_1 exited with code 0
```

skandragon, 2023-02-18:
After some searching, I found errors in my kernel log which indicated a USB power issue, most likely caused by some power saving feature.  I have disabled this in the kernel, and if the problem of crashing goes away, I will update this issue with the fix and symptom, for adding to your FAQ.

skandragon, 2023-02-19:
No luck.  Tried multiple USB cables.  I added reporting of the error reported by the hackrf library call, and it was `-1000` meaning libusb error.

shajen, 2023-02-19:
This message is caused directly by `libhackrf` library. Can you confirm that you `HackRf` device is working with other software correctly for a long time e.g `gqrx`, `sdr++`?

skandragon, 2023-03-13:
Indeed it works for extended amounts of time with sdr++ and gqrx, sdrangel, etc.  Extended is at least a few hours, often 4 or more.

syehorov, 2024-01-15:
Hello.
I faced the same issue and found interesting thing. I have device made by Great Scott Gadgets, and it may work fine for days, but may fail within seconds after scanner started.

>Board ID Number: 2 (HackRF One)
>Firmware Version: 2023.01.1 (API:1.07)
>Part ID Number: 0xa000cb3c 0x004f4762

Spare one, with no firmware updates, works the same bad:
>Board ID Number: 2 (HackRF One)
>Firmware Version: 2022.09.1 (API:1.06)
>Part ID Number: 0xa000cb3c 0x00744f6d

At the same time, friend of mine have a different hardware (aluminum case and labeled not as GSG), and it is working perfectly.

> Board ID Number: 2 (HackRF One)
> Firmware Version: 2021.03.1 (API:1.04)
> Part ID Number: 0xa000cb3c 0x0061475e

As you may see, it have different firmware and, maybe, it is a root cause. I`ll try do downgrade one of my devices, and see if it will help.

syehorov, 2024-01-16:
Update: I was wrong. It was only for a first few hours. Now it is the same issues.

syehorov, 2024-01-18:
Logs from devel version, latest docker image, amd64 arch.
```[2024-01-18 14:29:24.990903] [info   ] [main       ] git commit: 4d83913754de5415327042fb389cde5f3178132b
[2024-01-18 14:29:24.991017] [info   ] [main       ] git tag: 
[2024-01-18 14:29:24.991030] [info   ] [main       ] build type: release
[2024-01-18 14:29:24.991043] [info   ] [config     ] mqtt: admin@broker:1883
[2024-01-18 14:29:24.991135] [info   ] [config     ] file: {"cores":2,"detection":{"frequency_grouping_size":5000,"frequency_range_scanning_time_ms":64,"noise_detection_margin":20,"noise_learning_time_seconds":20,"torn_transmission_learning_time_seconds":20},"ignored_frequencies":[],"memory_limit_mb":8192,"output":{"console_log_level":"trace","file_log_level":"warning","logs":"sdr/logs"},"recording":{"max_noise_time_ms":500,"min_sample_rate":16000,"min_time_ms":1000},"scanned_frequencies":[{"device_enabled":true,"device_gains":{"AMP":14,"LNA":40,"VGA":62},"device_serial":"0000000000000000f77c60dc273a50c3","ranges":[{"sample_rate":20480000,"start":130000000,"stop":170000000},{"sample_rate":20480000,"start":380000000,"stop":520000000}]},{"device_enabled":false,"device_gains":{"IFGR":59,"RFGR":9},"device_serial":"223503C899","ranges":[{"sample_rate":2048000,"start":144000000,"stop":146000000},{"sample_rate":2048000,"start":438000000,"stop":440000000}]}],"version":1}
[2024-01-18 14:29:24.991187] [info   ] [main       ] scanner id: e0e299e99bfe4e6c875f8535ddf8da06da06
[2024-01-18 14:29:24.991224] [info   ] [main       ] start app auto_sdr
[2024-01-18 14:29:24.991235] [info   ] [main       ] start thread id: 8
[2024-01-18 14:29:24.992358] [info   ] [Mqtt       ] start thread id: 12
[2024-01-18 14:29:24.997161] [info   ] [Mqtt       ] connected
[2024-01-18 14:29:25.566637] [info   ] [SoapySDR   ] found 2 devices:
[2024-01-18 14:29:25.566712] [info   ] [SoapySDR   ] #0, name: HackRF One, driver: hackrf, version: 2023.01.1, serial: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:26.102204] [info   ] [SoapySDR   ]  gain: LNA, min: 0 dB, max: 40 dB
[2024-01-18 14:29:26.102259] [info   ] [SoapySDR   ]  gain: AMP, min: 0 dB, max: 14 dB
[2024-01-18 14:29:26.102271] [info   ] [SoapySDR   ]  gain: VGA, min: 0 dB, max: 62 dB
[2024-01-18 14:29:26.102284] [info   ] [SoapySDR   ]  frequency range: 0 MHz - 7250 MHz 
[2024-01-18 14:29:26.102304] [info   ] [SoapySDR   ]  sample rates: 1 MHz, 2 MHz, 3 MHz, 4 MHz, 5 MHz, 6 MHz, 7 MHz, 8 MHz, 9 MHz, 10 MHz, 11 MHz, 12 MHz, 13 MHz, 14 MHz, 15 MHz, 16 MHz, 17 MHz, 18 MHz, 19 MHz, 20 MHz
[2024-01-18 14:29:26.613195] [info   ] [SoapySDR   ] #1, driver: sdrplay, serial: 223503C899
[2024-01-18 14:29:26.652759] [info   ] [SoapySDR   ]  gain: IFGR, min: 20 dB, max: 59 dB
[2024-01-18 14:29:26.652834] [info   ] [SoapySDR   ]  gain: RFGR, min: 0 dB, max: 9 dB
[2024-01-18 14:29:26.652857] [info   ] [SoapySDR   ]  frequency range: 0 MHz - 2000 MHz 
[2024-01-18 14:29:26.652891] [info   ] [SoapySDR   ]  sample rates: 62500, 96 kHz, 125 kHz, 192 kHz, 250 kHz, 384 kHz, 500 kHz, 768 kHz, 1 MHz, 2 MHz, 2048 kHz, 3 MHz, 4 MHz, 5 MHz, 6 MHz, 7 MHz, 8 MHz, 9 MHz, 10 MHz
[2024-01-18 14:29:28.287644] [info   ] [RingBuffer ] init, buffer size: 41943040
[2024-01-18 14:29:28.841202] [info   ] [SoapySDR   ] open device: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:28.841457] [info   ] [SoapySDR   ] set offset: 0, device: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:28.841514] [info   ] [SoapySDR   ] set gain AMP: 14, device: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:28.841529] [info   ] [SoapySDR   ] set gain LNA: 40, device: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:28.841770] [info   ] [SoapySDR   ] set gain VGA: 62, device: 0000000000000000f77c60dc273a50c3
[2024-01-18 14:29:28.841978] [info   ] [spectrogram] init
[2024-01-18 14:29:28.842105] [info   ] [spectrogram] init
[2024-01-18 14:29:28.842249] [info   ] [Scanner    ] original frequency ranges: 2
[2024-01-18 14:29:28.842283] [info   ] [Scanner    ] frequency range start: 130.000 kHz, stop: 170.000 kHz, sample rate: 020.480 kHz, fft: 0
[2024-01-18 14:29:28.842301] [info   ] [Scanner    ] frequency range start: 380.000 kHz, stop: 520.000 kHz, sample rate: 020.480 kHz, fft: 0
[2024-01-18 14:29:28.842369] [info   ] [SamplesWrk ] start thread id: 56
[2024-01-18 14:29:28.842382] [info   ] [Scanner    ] splitted frequency ranges: 9
[2024-01-18 14:29:28.842413] [info   ] [SamplesWrk ] start thread id: 57
[2024-01-18 14:29:28.842635] [info   ] [Scanner    ] frequency range, start: 130.000 kHz, stop: 150.000 kHz, fft: 16384
[2024-01-18 14:29:28.842780] [info   ] [Scanner    ] frequency range, start: 150.000 kHz, stop: 170.000 kHz, fft: 16384
[2024-01-18 14:29:28.842803] [info   ] [Scanner    ] frequency range, start: 380.000 kHz, stop: 400.000 kHz, fft: 16384
[2024-01-18 14:29:28.842818] [info   ] [Scanner    ] frequency range, start: 400.000 kHz, stop: 420.000 kHz, fft: 16384
[2024-01-18 14:29:28.842834] [info   ] [Scanner    ] frequency range, start: 420.000 kHz, stop: 440.000 kHz, fft: 16384
[2024-01-18 14:29:28.842874] [info   ] [Scanner    ] frequency range, start: 440.000 kHz, stop: 460.000 kHz, fft: 16384
[2024-01-18 14:29:28.842889] [info   ] [Scanner    ] frequency range, start: 460.000 kHz, stop: 480.000 kHz, fft: 16384
[2024-01-18 14:29:28.842905] [info   ] [Scanner    ] frequency range, start: 480.000 kHz, stop: 500.000 kHz, fft: 16384
[... cut ...]
[2024-01-18 14:29:34.754544] [error  ] [main       ] some device stop working
[2024-01-18 14:29:34.754717] [info   ] [SamplesWrk ] stop thread id: 56
[2024-01-18 14:29:34.754965] [info   ] [spectrogram] deinit
[2024-01-18 14:29:34.754992] [info   ] [fft        ] deinit
[2024-01-18 14:29:34.755088] [info   ] [SamplesWrk ] stop thread id: 57
[2024-01-18 14:29:34.755352] [info   ] [spectrogram] deinit
[2024-01-18 14:29:34.755395] [info   ] [fft        ] deinit
[2024-01-18 14:29:35.416192] [info   ] [Mqtt       ] stop thread id: 12
```

syehorov, 2024-03-29:
Got it fixed for me when noticed CPU throttling in logs. Have CPU cooling solution upgraded and issue disappeared. About 2 months with no one issue.

shajen, 2025-10-20:
Check out the new version https://github.com/shajen/sdr-hub and let us know about the results.