shajen/rtl-sdr-scanner-cpp #52: Scanning Multiple Frequency Ranges

lonseidman, 2023-02-18:
Hi everyone,

Quick question - I noticed in the docs that it's possible to have the the scanner look at two different frequency ranges as it does its scanning. Per the docs:

`{
  "scanner_frequencies_ranges": [
    {
      "device_serial": "auto",
      "ranges": [
        {
          "start": 144000000,
          "stop": 146000000,
          "sample_rate": 2048000
        },
        {
          "start": 440000000,
          "stop": 442000000,
          "sample_rate": 2048000
        }
      ]
    }
  ]
}`

When I set this up in my installation I can see the application bouncing back and forth between the two frequency ranges but it's not recording anything - including strong signals that are normally recorded when I have only one range in my config file. Any thoughts or is this not supported with a single device?

skandragon, 2023-02-18:
It is supported, and apparently has or does work for others.  I am also unable to get any recordings, including strong signals.

Of course, I also don't get signals even if I use a single range...

skandragon, 2023-02-19:
I see this in my logs, but no transmissions are recorded:
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.629177] [info   ] [RecorderWrk] start thread id: 33, frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.734083] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.746841] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.767050] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.845834] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.910061] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:27.971589] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:28.037653] [warning] [DataCtrl   ] start transmission not active frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:28.076732] [info   ] [TornDtr    ] report transmission frequency: 469.950 kHz, duration: 0.00 seconds
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:28.077136] [warning] [DataCtrl   ] finish transmission not found frequency: 469.950 kHz
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:28.077168] [info   ] [RecorderWrk] stop thread id: 33, frequency: 469.950 kHz, queue size: 0
sdr-scanner-sdr-scanner-1  | [2023-02-18 20:11:28.077288] [info   ] [Recorder   ] erase worker frequency: 469.950 kHz, total workers: 0
```
```

jointerCZ, 2023-02-19:
It works with my 2 RTL.
One is scanning 2 bands, second 4 bands.
Runs stable for 2 days currently, however I found I have to have each connected via separate 2m USB cable, hub was not stable and powered hub was also not  stable and also creating noise in all spectrograms.
It currently runs in VMWare Player on my desktop.

Klammraute, 2023-02-19:
Its important to write the code correctly, took me about 1 month to figure it out. 
i can provide you my config file with 2 dongles and multiple frequencyranges

[config_sdrscanner_github.txt](https://github.com/shajen/rtl-sdr-scanner-cpp/files/10777701/config_sdrscanner_github.txt)

algorni, 2023-03-02:
What's the maximum frequency band you can configure at time? I saw normally chunk of 2MHz is this due to the SDR device limitation ,right?   this means if i wanna scan a wider range i need to configure multiple ranges and having the SW jumping between those ranges, right?