shajen/rtl-sdr-scanner-cpp #55: Error response from daemon: error gathering device information while adding custom device "/dev/bus/usb": no such file or directory

thekoos, 2023-02-23:
Get the following error when running "`docker compose up`":

`Error response from daemon: error gathering device information while adding custom device "/dev/bus/usb": no such file or directory`

### **Hackrf is connected and reports status via hackrf_info**

xxxxxxxx@xxxxxx:~/sdr$ hackrf_info 
hackrf_info version: unknown
libhackrf version: unknown (0.6)
Found HackRF
Index: 0
Serial number: 000000000000000014d463dc2f1351e1
Board ID Number: 2 (HackRF One)
Firmware Version: 2022.09.1 (API:1.06)
Part ID Number: 0xa000cb3c 0x0053434d
### 
**Full trace running docker compos up:**

kobus@Dozer:~/sdr$ docker compose up
[+] Running 6/6
 ⠿ Network sdr_default                 Created                                                                                                                                                        0.0s
 ⠿ Container sdr-sdr-broker-1          Created                                                                                                                                                        0.1s
 ⠿ Container sdr-sdr-monitor-setup-1   Created                                                                                                                                                        0.1s
 ⠿ Container sdr-sdr-scanner-1         Created                                                                                                                                                        0.1s
 ⠿ Container sdr-sdr-monitor-server-1  Created                                                                                                                                                        0.1s
 ⠿ Container sdr-sdr-monitor-reader-1  Created                                                                                                                                                        0.1s
Attaching to sdr-sdr-broker-1, sdr-sdr-monitor-reader-1, sdr-sdr-monitor-server-1, sdr-sdr-monitor-setup-1, sdr-sdr-scanner-1
sdr-sdr-broker-1          | 1677113658: mosquitto version 2.0.15 starting
sdr-sdr-broker-1          | 1677113658: Config loaded from /mosquitto/config/mosquitto.conf.
sdr-sdr-broker-1          | 1677113658: Opening ipv4 listen socket on port 1883.
sdr-sdr-broker-1          | 1677113658: Opening ipv6 listen socket on port 1883.
sdr-sdr-broker-1          | 1677113658: mosquitto version 2.0.15 running
sdr-sdr-monitor-setup-1   | timezone: Etc/UTC
sdr-sdr-monitor-setup-1   | Operations to perform:
sdr-sdr-monitor-setup-1   |   Apply all migrations: admin, auth, contenttypes, graphs, logs, sdr, sessions
sdr-sdr-monitor-setup-1   | Running migrations:
sdr-sdr-monitor-setup-1   |   No migrations to apply.
sdr-sdr-monitor-setup-1 exited with code 0
Error response from daemon: error gathering device information while adding custom device "/dev/bus/usb": no such file or directory

Am I missing a configuration step?

Thank you for all your great work on this tool.

Xecsc, 2023-08-08:
Struggled with this for a day. 
Simple just need to run 'sudo docker compose up' 
Probably should delete add images and volumes first. 

Hopes that helps anyone else whos stuck on that error.

jeremyfuksa, 2023-10-25:
I'm also having this problem. Docker Desktop 4.24.2 on macOS. Same output as above, except I am using an RTL-SDR dongle. Running as sudo did not help, even after a clean pull and spin-up.

jeremyfuksa, 2023-10-27:
I've isolated the problem to the dongle being attached to the kernel. But the blacklist instructions only work for Linux, as modprobe is not available on macOS.