shajen/rtl-sdr-scanner-cpp #57: Pruning spectrograms

great9, 2023-02-26:
Is there a way to automatically prune spectrograms that are older than x hours or x days?

algorni, 2023-02-27:
similar question for recording?

great9, 2023-02-27:
> similar question for recording?

Recordings should have an option to keep a recording.

shajen, 2023-02-28:
There is a mechanism to delete recent transmissions over `10 GB`. A similar mechanism is for spectrograms.

You can change the size of the quota in the `docker-compose.yml`.

Change line:
```
command: python3 manage.py runscript mqtt_reader
```
to
```
command: python3 manage.py runscript mqtt_reader sdr-broker 1883 admin password 10 10
```
Where two last digits mean size in GB of transmissions and spectrograms.

Hope it helps you.

great9, 2023-02-28:
That's the docker volume mechanism. Why not implement this functionality as
a cronjob ?

shajen, 2023-03-01:
What do you mean? Native `cronjob` on host machine?

It's not good idea because everyone have to configure it separately in his OS. Docker version works out of the box without any changes in host os. Moreover, it's hard to access django's files in docker directly from the host OS.

great9, 2023-03-04:
cronjob as in a script that deletes spectrograms and recordings from the docker volumes

upya4ko, 2023-03-04:
> to
> 
> ```
> command: python3 manage.py runscript mqtt_reader sdr-broker 1883 admin password 10 10
> ```

Hi i try to extend storage limit to temporary skip issue #60 , in some reason container fail to start with this error:
```
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | CommandError: An error has occurred running scripts. See errors above.
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | timezone: UTC
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script 'sdr-broker' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script '1883' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script 'admin' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script 'password' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script '40' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | No (valid) module for script '40' found
rtl-sdr-scanner-cpp-sdr-monitor-reader-1  | Try running with a higher verbosity level like: -v2 or -v3
rtl-sdr-scanner-cpp-sdr-monitor-reader-1 exited with code 0
```

In docker-compose.yml i put:
```
command: python3 manage.py runscript mqtt_reader sdr-broker 1883 admin password 40 40
```
instead of 
```
command: python3 manage.py runscript mqtt_reader
```

Can you tell please what i doing wrong?

shajen, 2023-03-04:
> cronjob as in a script that deletes spectrograms and recordings from the docker volumes. it can work as a docker cronjob, can be fired off by django not the os itself

I know what is a cronjob. How do you imagine deleting recordings using cronjob? To delete recordings you have to delete some data in db, not only files on disk.

> Can you tell please what i doing wrong?

My mistake. The correct command is:
```
python3 manage.py runscript mqtt_reader --script-args sdr-broker 1883 admin password 40 40
```

great9, 2023-03-04:
> I know what is a cronjob. How do you imagine deleting recordings using cronjob? To delete recordings you have to delete some data in db, not only files on disk.

so why deleting db entries from django which already has access to the db is a problem? maybe im not following, I haven't looked at that code

algorni, 2023-03-06:
is there also a way to lower down the logs level of the manage.py script by command line?