shajen/rtl-sdr-scanner-cpp #61: Windows Support

PierrunoYT, 2023-03-13:
Can you add Windows Support?

dziban303, 2023-04-27:
It works on Windows. Use Docker Desktop in conjunction with WSL2. Install [usbipd](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) to give access to your USB bus to WSL2 and Docker. Then follow the instruction in the repo.

hmdsamarasekara, 2023-08-13:
Sorry for asking a noob question but can you please provide instructions on how to configure usbipd after installation. I'm having trouble getting it done. when I run "usbipd wsl attach --busid 1-2" I get this error "The selected WSL distribution is not running; keep a command prompt to the distribution open to leave it running."

hmdsamarasekara, 2023-08-13:
[image]

PLEASE HELP...!!!

RobertDWhite, 2023-08-13:
Remove "WSL" from the command, and try again.

hmdsamarasekara, 2023-08-13:
[image]

Not working, this error comes when i run without wsl...

dziban303, 2023-08-13:
`wsl` needs to be in there but it's trying to attach to a instance that isn't running, like it says in the error message. Try starting Docker so that its WSL instance starts. If that doesn't work you need to attach it to a different distro. 

Run `wsl --list` which will show you what distros you have installed. 
[image]

Then you can run `usbipd wsl attach -b 1-2 -d Ubuntu` or whatever to bind to that instance. You will, of course, need it running first.

If you don't have anything but docker-desktop and docker-desktop-data you may need to install, for example, Ubuntu. I don't know, I've never tried running WSL without a real linux distro installed as that's kind of the point of WSL.

hmdsamarasekara, 2023-08-13:
I have little knowledge when it comes to Linux and docker, what I did was just installed docker and searched for "shajen/sdr-scanner" within docker, download it, and ran it. then try to figure out "usbipd"...
Do I need to download a Linux distro as well?

hmdsamarasekara, 2023-08-13:
[image]

As you said I only have these two entries...

dziban303, 2023-08-13:
Like I said, I don't know. This is the blind leading the blind, I really can't offer any more suggestions, sorry. Good luck

RobertDWhite, 2023-08-13:
Yeah I didn't realize you were using CMD since I wasn't paying attention. Try launching the WSL Ubuntu Terminal instead of CMD and try there. That's usually how I use WSL.

shajen, 2025-10-20:
New release should be much easier to launch on windows. Check out the new version https://github.com/shajen/sdr-hub