shajen/rtl-sdr-scanner-cpp #73: I'm not sure what is happening but recording transmission isn't working

MaxDeepCoder, 2023-07-05:
`sdr-sdr-monitor-setup-1 exited with code 0`

MaxDeepCoder, 2023-07-18:
The default one. Didn't change it or anything

Alifalisda, 2023-07-31:
hello, i have the same issues. This is my config @RobertDWhite.
{
  "scanner_frequencies_ranges": [
    {
      "device_serial": "auto",
      "ranges": [
       {
          "start": 102000000,
          "stop": 103000000,
          "step": 1000,
          "sample_rate": 2048000
        },
        {
          "start": 144000000,
          "stop": 146000000,
          "step": 1000,
          "sample_rate": 2048000
        }
      ]
    }
  ],
  "ignored_frequencies": [],
  "devices": {
    "rtl_sdr": {
      "ppm_error": -2.43,
      "tuner_gain": 50,
      "offset": 0
    },
    "hack_rf": {
      "lna_gain": 0,
      "vga_gain": 0,
      "offset": 0
    }
  },
  "recording": {
    "max_noise_time_ms": 500,
    "min_time_ms": 2000,
    "min_sample_rate": 16000
  },
  "detection": {
    "frequency_grouping_size": 10000,
    "frequency_range_scanning_time_ms": 64,
    "noise_learning_time_seconds": 30,
    "noise_detection_margin": 10,
    "torn_transmission_learning_time_seconds": 60
  },
  "output": {
    "logs": "sdr/logs",
    "file_log_level": "info",
    "console_log_level": "info"
  },
  "cores": 4,
  "memory_limit_mb": 0,
  "mqtt": {
    "hostname": "sdr-broker",
    "port": 1883,
    "username": "admin",
    "password": "password"
  }
}

WickedxJosh, 2024-05-29:
Was there ever a fix for this? I keep getting the same error, sdr-monitor-1 exited with code 0

so1der, 2024-11-10:
Same error. Arch Linux. Scanner was installed via docker, default config, RTL-SDR V4.