shajen/rtl-sdr-scanner-cpp #75: Compile from Source yields cmake errors

Zotter, 2023-08-07:
Fedora 38, up to date and pre-reqs installed

Following build instructions - all goes well until:

`[ 72%] Building CXX object CMakeFiles/auto_sdr_libs.dir/sources/ring_buffer.cpp.o
[ 75%] Building CXX object CMakeFiles/auto_sdr_libs.dir/sources/utils.cpp.o
/home/<user>/sdr-scanner/sources/utils.cpp:76:3: error: ‘f32’ or ‘F32’ suffix on floating constant only available with ‘-std=c++2b’ or ‘-std=gnu++2b’ [-Werror=pedantic]
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |   ^~~~~
/home/<user>/sdr-scanner/sources/utils.cpp: In function ‘std::vector<std::complex<float> > getShiftData(int32_t, Frequency, uint32_t)’:
/home/<user>/sdr-scanner/sources/utils.cpp:76:58: error: no match for ‘operator*’ (operand types are ‘std::complex<float>’ and ‘_Float32’)
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
      |                                                   |
      |                                                   std::complex<float>
In file included from /usr/include/c++/13/valarray:605,
                 from /usr/include/nlohmann/detail/conversions/from_json.hpp:21,
                 from /usr/include/nlohmann/adl_serializer.hpp:14,
                 from /usr/include/nlohmann/json.hpp:35,
                 from /home/<user>/sdr-scanner/sources/config.h:6,
                 from /home/<user>/sdr-scanner/sources/utils.h:3,
                 from /home/<user>/sdr-scanner/sources/utils.cpp:1:
/usr/include/c++/13/bits/valarray_after.h:407:5: note: candidate: ‘template<class _Dom1, class _Dom2> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_Expr, std::_Expr, _Dom1, _Dom2>, typename std::__fun<std::__multiplies, typename _Dom1::value_type>::result_type> std::operator*(const _Expr<_Dom1, typename _Dom1::value_type>&, const _Expr<_Dom2, typename _Dom2::value_type>&)’
  407 |     _DEFINE_EXPR_BINARY_OPERATOR(*, struct std::__multiplies)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note:   template argument deduction/substitution failed:
In file included from /usr/include/bits/floatn.h:119,
                 from /usr/include/wchar.h:30,
                 from /usr/include/c++/13/cwchar:44,
                 from /usr/include/c++/13/bits/postypes.h:40,
                 from /usr/include/c++/13/bits/char_traits.h:42,
                 from /usr/include/c++/13/string:42,
                 from /home/<user>/sdr-scanner/sources/radio/help_structures.h:4,
                 from /home/<user>/sdr-scanner/sources/config.h:3:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   ‘std::complex<float>’ is not derived from ‘const std::_Expr<_Dom1, typename _Dom1::value_type>’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note: candidate: ‘template<class _Dom> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_Expr, std::_Constant, _Dom, typename _Dom::value_type>, typename std::__fun<std::__multiplies, typename _Dom1::value_type>::result_type> std::operator*(const _Expr<_Dom1, typename _Dom1::value_type>&, const typename _Dom::value_type&)’
  407 |     _DEFINE_EXPR_BINARY_OPERATOR(*, struct std::__multiplies)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   ‘std::complex<float>’ is not derived from ‘const std::_Expr<_Dom1, typename _Dom1::value_type>’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note: candidate: ‘template<class _Dom> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_Constant, std::_Expr, typename _Dom::value_type, _Dom>, typename std::__fun<std::__multiplies, typename _Dom1::value_type>::result_type> std::operator*(const typename _Dom::value_type&, const _Expr<_Dom1, typename _Dom1::value_type>&)’
  407 |     _DEFINE_EXPR_BINARY_OPERATOR(*, struct std::__multiplies)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   mismatched types ‘const std::_Expr<_Dom1, typename _Dom1::value_type>’ and ‘_Float32’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note: candidate: ‘template<class _Dom> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_Expr, std::_ValArray, _Dom, typename _Dom::value_type>, typename std::__fun<std::__multiplies, typename _Dom1::value_type>::result_type> std::operator*(const _Expr<_Dom1, typename _Dom1::value_type>&, const valarray<typename _Dom::value_type>&)’
  407 |     _DEFINE_EXPR_BINARY_OPERATOR(*, struct std::__multiplies)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   ‘std::complex<float>’ is not derived from ‘const std::_Expr<_Dom1, typename _Dom1::value_type>’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note: candidate: ‘template<class _Dom> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_ValArray, std::_Expr, typename _Dom::value_type, _Dom>, typename std::__fun<std::__multiplies, typename _Dom1::value_type>::result_type> std::operator*(const valarray<typename _Dom::value_type>&, const _Expr<_Dom1, typename _Dom1::value_type>&)’
  407 |     _DEFINE_EXPR_BINARY_OPERATOR(*, struct std::__multiplies)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/valarray_after.h:407:5: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   mismatched types ‘const std::_Expr<_Dom1, typename _Dom1::value_type>’ and ‘_Float32’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/valarray:1192:1: note: candidate: ‘template<class _Tp> std::_Expr<std::__detail::_BinClos<std::__multiplies, std::_ValArray, std::_ValArray, _Tp, _Tp>, typename std::__fun<std::__multiplies, _Tp>::result_type> std::operator*(const valarray<_Tp>&, const valarray<_Tp>&)’
 1192 | _DEFINE_BINARY_OPERATOR(*, __multiplies)
      | ^~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/valarray:1192:1: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   ‘std::complex<float>’ is not derived from ‘const std::valarray<_Tp>’
[... 19 lines cut ...]
      |     ^~~~~~~~
/usr/include/c++/13/complex:395:5: note:   template argument deduction/substitution failed:
/home/<user>/sdr-scanner/sources/utils.cpp:76:60: note:   mismatched types ‘const std::complex<_Tp>’ and ‘_Float32’
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
/usr/include/c++/13/complex:404:5: note: candidate: ‘template<class _Tp> std::complex<_Tp> std::operator*(const complex<_Tp>&, const _Tp&)’
  404 |     operator*(const complex<_Tp>& __x, const _Tp& __y)
      |     ^~~~~~~~
/usr/include/c++/13/complex:404:5: note:   template argument deduction/substitution failed:
[... cut ...]
   76 |   const auto f = std::complex<float>(0.0f, -1.0f) * 2.0f * M_PIf32 * (static_cast<float>(-frequencyOffset) / static_cast<float>(sampleRate));
      |                                                            ^~~~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/auto_sdr_libs.dir/build.make:412: CMakeFiles/auto_sdr_libs.dir/sources/utils.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/auto_sdr_libs.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
`

aqbifzl, 2023-09-01:
I had very similar issue. Try after my pull request (https://github.com/shajen/rtl-sdr-scanner-cpp/pull/77) and lmk if it also resolved your problem

Zotter, 2023-10-24:
In my case, I solved things by setting up a dedicated Debian box  and installed via Docker - it's now working and not on my main machine - turned out for the better after all. Thanks!

closed 2023-10-24 by Zotter