shajen/rtl-sdr-scanner-cpp #76: Stale?

tubalainen, 2023-08-26:
Hi @shajen !

First of all - thanks for a fantastic software! Great functionality!
With all respect, this is open source and you are doing this in your private/own time. Other things in the world can be more important than maintaining a git repo.

So, a very humble, as humble as this question can be. Are there any plans to continue to improve and evolve this repo?

Best regards,
Tubalainen

shajen, 2023-08-28:
Thank you for the very kind words!

The project is still in progress.  I have a huge update planned over the next few weeks/months, including: 
- classification of recordings using artificial intelligence (detect noise, speech etc.)
- web configuration
- using SoapySDR library that support almost all sdr device
- recordings grouping
- publication of web panel sources
- and more

I need time to test it all, which I haven't had much lately, so please be patient.

tubalainen, 2023-09-27:
Hi @shajen please shout if I want some help with testing and eval. :) Hope development is going well.

shajen, 2023-10-31:
Hello,
the final version is almost ready! The **development** version is available for test, follow these steps to run:

**IMPORTANT**

- old config is dropped, you have to configure it again via [gui](http://localhost:8000/sdr/config), but now it's very easy, check [manual](https://github.com/shajen/rtl-sdr-scanner-cpp/wiki/Configuration)

- `transmissions` and `spectrograms` are transferred but you have to use the same directory for docker compose (due to default docker compose volume name) which was used in previous version (`sdr` in home directory by default)

stop and remove old version:
```
cd ~/sdr
docker compose down
cd ..
rm -rf ~/sdr
```

download and run new version
```
git clone --branch devel https://github.com/shajen/rtl-sdr-scanner-cpp.git ~/sdr
cd ~/sdr
sed -i 's/latest/devel/' .env
nano .env # customize settings if you want: timezone, admin account and other
docker compose up -d
```

More info in [README](https://github.com/shajen/rtl-sdr-scanner-cpp/tree/devel) on devel branch.

I hope it works! Enjoy it!

ar8001, 2023-11-01:
new version  crashs here: 

sdr-sdr-monitor-setup-1 exited with code 0
Error response from daemon: driver failed programming external connectivity on endpoint sdr-sdr-monitor-server-1 (33b5118a36517a3e5aee040cb2dc149ea6e1df26e657737e079fe0ef5c1a3818): Bind for 0.0.0.0:8000 failed: port is already allocated
root@openwebrx:/home/mike/sdr#

tubalainen, 2023-11-01:
> new version crashs here:
> 
> sdr-sdr-monitor-setup-1 exited with code 0 Error response from daemon: driver failed programming external connectivity on endpoint sdr-sdr-monitor-server-1 (33b5118a36517a3e5aee040cb2dc149ea6e1df26e657737e079fe0ef5c1a3818): Bind for 0.0.0.0:8000 failed: port is already allocated root@openwebrx:/home/mike/sdr#
> 
> ps. i send u a smal donation for your work! keep it up please!!

Please stop your old sdrscanner instance. Some other docker/service is utilizing port 8000 .

:)

tubalainen, 2023-11-01:
@shajen WAAAHOOO!! Thanks a billion!

I am using my attic server for many things, AIS-catcher, dump1090. They are hugging RTL/Airspy devices for each service.
Ive got two devices allocated for sdrscanner. One airspy r2 and one rtl-sdr.

How ever - hence the other devices on the machine are used by other services sdrscanner cannot connect to them and utilize them the system exits and no devices are presented to sdrscanner.

```
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.838612] [info   ] [SoapySDR   ] found 4 devices:
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.838634] [info   ] [SoapySDR   ] #0, driver: airspy, serial: 62cc68ff3584805b
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.848847] [info   ] [SoapySDR   ]  gain: LNA, min: 0 dB, max: 15 dB
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.848871] [info   ] [SoapySDR   ]  gain: MIX, min: 0 dB, max: 15 dB
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.848875] [info   ] [SoapySDR   ]  gain: VGA, min: 0 dB, max: 15 dB
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.848880] [info   ] [SoapySDR   ]  frequency range: 24 MHz - 1800 MHz
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.848888] [info   ] [SoapySDR   ]  sample rates: 10 MHz, 2500 kHz
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:31.849232] [info   ] [SoapySDR   ] #1, name: Realtek RTL2838UHIDIR unavailable, driver: rtlsdr, serial: ais
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:32.006904] [info   ] [Mqtt       ] stop thread id: 10
sdrscanner-broker-1          | 1698829292: Client auto-346B48ED-E70A-773F-D8DF-1B016D3894A7 disconnected.
sdrscanner-sdr-scanner-1     | [2023-11-01 10:01:32.007568] [error  ] [main       ] main exception: Unable to open RTL-SDR device
sdrscanner-sdr-scanner-1 exited with code 1

```
If i stop dump1090 and ais-catcher and free up all connected devices and then start sdrscanner it all works like a charm.

Thanks.

ar8001, 2023-11-02:
I have now installed the denvelop version and am thrilled again! You're doing a great job!

I now have a nooelec dongle and an AirSpy Mini connected. But what I noticed is that when I use the AirSpy mini and set the sample rate to 8 MHz, the audio files are much too fast. They sound as if they are being played at 4x speed.

shajen, 2023-11-03:
@tubalainen 

> I am using my attic server for many things, AIS-catcher, dump1090. They are hugging RTL/Airspy devices for each service. Ive got two devices allocated for sdrscanner. One airspy r2 and one rtl-sdr.
> 
> How ever - hence the other devices on the machine are used by other services sdrscanner cannot connect to them and utilize them the system exits and no devices are presented to sdrscanner.
> 
> If i stop dump1090 and ais-catcher and free up all connected devices and then start sdrscanner it all works like a charm.

It's a bug. Thank you for investigation and very useful debug info! I just pushed a fix. Please download latest images, run again and check if it work:
```
docker compose pull 
docker compose up
```

@ar8001 

> ps. i send u a smal donation for your work! keep it up please!!

Thank you! I got it.

> I now have a nooelec dongle and an AirSpy Mini connected. But what I noticed is that when I use the AirSpy mini and set the sample rate to 8 MHz, the audio files are much too fast. They sound as if they are being played at 4x speed. The files recordet with the nooelec and 2MHz Sample mode are normal
> 
> if i change the samplerate of the airspy mini back to 2 MHz .. the audiofiles are to slow, sounds like 0.25 speed

I have no `AirSpy Mini` - I have only `rtl-sdr` and `HackRf`. It is really hard to fix a bug without hardware.
Please install `SoapySDR` (`sudo apt-get install soapysdr-tools`) on your host machine and send back full output of command (run for each your devices separately) while `sdr-scanner` is down:
```
SoapySDRUtil --probe=serial=XXX
```
where `xxx` is a serial of device (to find serial run `SoapySDRUtil --find`).

JulianDFarmer, 2023-11-04:
Hi,

I have an SDRPlay RSP1A and unfortunately can't get it working with the devel docker image. I think it's due to specific drivers / dependencies being required to get it working with SoapySDR. For example, the drivers located here:
https://www.sdrplay.com/downloads/

I've managed to get the RSP1A working on my machine with other SDR tools (running Debian) using the instructions here:
https://groups.io/g/SDRPlayUsers/topic/83829243

But I don't know enough about docker to get your image to install these drivers when running docker compose.

If it helps, you can see how openwebrx handle it here:
https://github.com/jketterl/openwebrx/blob/develop/docker/scripts/install-dependencies-sdrplay.sh

Is there any scope to add SDRPlay support to the image, or do you have any thoughts on how to run this outside of docker as I have all the drivers installed and working on the host machine?

Many thanks!

tubalainen, 2023-11-05:
Thanks @shajen the device lock thing works great now! Only adapts and starts the devices that are not used! :)

**New bugs/Feature Req´s**
1.) Ive got a airspy r2 that is capable of 10MHz bandwidth. All "transmissions" are very distorted and sound very very odd. If I fall back to have 2.4MHz bandwidth it works. I am running the docker on a capable machine with a core i5 7500 processor and 32 gigs of memory. The CPU is idling. I am able to connect to the Airspy r2 via spyserver or soapsdrserver without any issues. So there is not a issue with the USB bus either. 
[image]
[image]
[image]
2.) When editing my second device I cannot push "save". When I do so it all hangs. All I did was removing the 70 cm ham band section. Any changes I make here will end up hanging up the system require me to "docker compose restart".

3.) Would it be possible not to start the whole system (including the new AI stuff for SnR etc) before you have set all config? Hence if you make a lot of changes the Signal to noise (for instance) will change. Can be gain, can be freq band and sections etc.
3A.) So, Would it be possible to introduce a "welcome to sdrscanner" wizard and after you have set it all up kick off the auto adjustments and the AI stuff?
3B,) Do the same, relearn SnR etc. when you make changes to your setup (even if it has been running for awhile)?

shajen, 2023-11-05:
@JulianDFarmer 

> I have an SDRPlay RSP1A and unfortunately can't get it working with the devel docker image. I think it's due to specific drivers / dependencies being required to get it working with SoapySDR.
>
> Is there any scope to add SDRPlay support to the image, or do you have any thoughts on how to run this outside of docker as I have all the drivers installed and working on the host machine?

Hi,

I added `SDRplay` support to docker image in 5432b24.

Save `udev` rules on your **host OS** to `/etc/udev/rules.d/66-mirics.rules` (if don't have yet):
```
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="2500",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="3000",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="3010",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="3020",MODE:="0666"
SUBSYSTEM=="usb",ENV{DEVTYPE}=="usb_device",ATTRS{idVendor}=="1df7",ATTRS{idProduct}=="3030",MODE:="0666"
```

Then reload `udev` rules:
```
sudo udevadm control --reload-rules && sudo udevadm trigger
```

Then run new images and check:
```
docker compose pull
docker compose up -d
```

shajen, 2023-11-05:
@tubalainen 

> **New bugs/Feature Req´s** 1.) Ive got a airspy r2 that is capable of 10MHz bandwidth. All "transmissions" are very distorted and sound very very odd. If I fall back to have 2.4MHz bandwidth it [...]

Try to set `10240000` Hz sample rate. It's very important that sample rate should be the power of 2 multiplied by some factor, 10240000 = 2^10 * 10000, so looks ok.

> 2.) When editing my second device I cannot push "save". When I do so it all hangs. All I did was removing the 70 cm ham band section. Any changes I make here will end up hanging up the system require me to "docker compose restart".

When you click save `sdr-scaner` module need restart, so you have to wait few seconds before click save again or refresh page. Moreover, you can change the settings of any devices and click save once - all settings will be saved.

> 3.) Would it be possible not to start the whole system (including the new AI stuff for SnR etc) before you have set all config? Hence if you make a lot of changes the Signal to noise (for instance) will change. Can be gain, can be freq band and sections etc.

Yes, you can do this by start only selected services (but then spectrograms and transmissions are not captured), set config, and run rest services:
```
docker compose up -d broker sdr-scanner monitor-server
#set config 
docker compose up -d
```

> 3A.) So, Would it be possible to introduce a "welcome to sdrscanner" wizard and after you have set it all up kick off the auto adjustments and the AI stuff?

Maybe in the distant future. I have some other features to do.

> 3B,) Do the same, relearn SnR etc. when you make changes to your setup (even if it has been running for awhile)?

It works as you wrote. If you change the configuration, the scanner learns the SnR again.

JulianDFarmer, 2023-11-05:
> Hi,
> 
> I added `SDRplay` support to docker image in [5432b24](https://github.com/shajen/rtl-sdr-scanner-cpp/commit/5432b247946cc31baf3035a39d1d519e9420ae7b).

Thanks, that works! I'm coming across an issue, though, where if I have more than one set of frequencies set to scan it does not record anything and resets itself every few minutes.

Let me know if you want me to open this as a separate issue - not sure if you would prefer to keep all the development branch issues separate from the main issues?

Here is an example of the logging output:
`sdr-sdr-scanner-1     | [2023-11-05 21:31:05.642236] [info   ] [Mqtt       ] connected
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.757485] [info   ] [SoapySDR   ] found 1 devices:
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.757540] [info   ] [SoapySDR   ] #0, driver: sdrplay, serial: 2235023799
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.798098] [info   ] [SoapySDR   ]  gain: IFGR, min: 20 dB, max: 59 dB
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.798133] [info   ] [SoapySDR   ]  gain: RFGR, min: 0 dB, max: 9 dB
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.798147] [info   ] [SoapySDR   ]  frequency range: 0 MHz - 2000 MHz 
sdr-sdr-scanner-1     | [2023-11-05 21:31:05.798169] [info   ] [SoapySDR   ]  sample rates: 62500, 96 kHz, 125 kHz, 192 kHz, 250 kHz, 384 kHz, 500 kHz, 768 kHz, 1 MHz, 2 MHz, 2048 kHz, 3 MHz, 4 MHz, 5 MHz, 6 MHz, 7 MHz, 8 MHz, 9 MHz, 10 MHz
sdr-sdr-scanner-1     | [2023-11-05 21:31:06.934939] [info   ] [RingBuffer ] init, buffer size: 41943040
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064016] [info   ] [SoapySDR   ] open device: 2235023799
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064211] [info   ] [SoapySDR   ] set offset: 0, device: 2235023799
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064230] [info   ] [SoapySDR   ] set gain IFGR: 40, device: 2235023799
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064237] [info   ] [SoapySDR   ] set gain RFGR: 2, device: 2235023799
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064273] [info   ] [spectrogram] init
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064353] [info   ] [spectrogram] init
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064458] [info   ] [spectrogram] init
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064524] [info   ] [SamplesWrk ] start thread id: 62
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064575] [info   ] [spectrogram] init
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064676] [info   ] [SamplesWrk ] start thread id: 63
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.064828] [info   ] [SamplesWrk ] start thread id: 64
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065120] [info   ] [Scanner    ] original frequency ranges: 2
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065145] [info   ] [Scanner    ] frequency range start: 130.000 kHz, stop: 135.000 kHz, sample rate: 005.120 kHz, fft: 0
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065153] [info   ] [Scanner    ] frequency range start: 155.500 kHz, stop: 157.500 kHz, sample rate: 002.048 kHz, fft: 0
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065155] [info   ] [SamplesWrk ] start thread id: 65
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065160] [info   ] [Scanner    ] splitted frequency ranges: 2
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065201] [info   ] [Scanner    ] frequency range, start: 130.000 kHz, stop: 135.000 kHz, fft: 4096
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065208] [info   ] [Scanner    ] frequency range, start: 155.500 kHz, stop: 157.500 kHz, fft: 2048
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.065349] [info   ] [Scanner    ] start thread id: 66
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.412950] [info   ] [fft        ] init size: 4096
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.413203] [info   ] [fft        ] init size: 4096
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.413207] [info   ] [fft        ] init size: 4096
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.413462] [info   ] [fft        ] init size: 4096
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.414586] [warning] [NoiseLrn   ] signals and noise do not match
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.414635] [info   ] [NoiseLrn   ] initialize, start: 130.000 kHz, stop: 135.000 kHz
sdr-sdr-scanner-1     | [2023-11-05 21:31:07.415260] [info   ] [SignalMediator] aggregation time: 1000 ms
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.145637] [info   ] [fft        ] init size: 2048
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.145656] [info   ] [fft        ] init size: 2048
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.146171] [info   ] [fft        ] init size: 2048
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.146218] [info   ] [fft        ] init size: 2048
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.146631] [warning] [NoiseLrn   ] signals and noise do not match
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.146643] [info   ] [NoiseLrn   ] initialize, start: 155.500 kHz, stop: 157.500 kHz
sdr-sdr-scanner-1     | [2023-11-05 21:31:09.146928] [info   ] [SignalMediator] aggregation time: 1000 ms
sdr-sdr-scanner-1     | [2023-11-05 21:33:44.444820] [info   ] [SdrDevice  ] average samples time: 1575.54 ms
sdr-sdr-scanner-1     | [2023-11-05 21:33:44.445114] [info   ] [Scanner    ] average samples time: 1573.80 ms
sdr-sdr-scanner-1     | [2023-11-05 21:36:22.872358] [info   ] [SdrDevice  ] average samples time: 1584.28 ms
sdr-sdr-scanner-1     | [2023-11-05 21:36:22.872649] [info   ] [Scanner    ] average samples time: 1584.27 ms
[... cut ...]
sdr-sdr-scanner-1     | [2023-11-05 21:50:40.852826] [info   ] [SamplesWrk ] stop thread id: 65
sdr-sdr-scanner-1     | [2023-11-05 21:50:40.853320] [info   ] [spectrogram] deinit
sdr-sdr-scanner-1     | [2023-11-05 21:50:40.853346] [info   ] [fft        ] deinit
sdr-sdr-scanner-1     | [2023-11-05 21:50:40.853360] [info   ] [fft        ] deinit
sdr-sdr-scanner-1     | [2023-11-05 21:50:41.406927] [info   ] [Mqtt       ] stop thread id: 12
sdr-broker-1          | 1699221041: Client auto-6261C31B-5097-C4EE-0B1A-7A967FD3F9F0 disconnected.
sdr-sdr-scanner-1 exited with code 0`

ar8001, 2023-11-06:
@shajen  : if you need access to a remote system with AirSpy Mini .... let me know,  and i will send you more donations from time to time to support you and this project

shajen, 2023-11-06:
@JulianDFarmer

> Thanks, that works! I'm coming across an issue, though, where if I have more than one set of frequencies set to scan it does not record anything and resets itself every few minutes.
>
> Let me know if [...]

This is problem with `SoapySdr`. I'm working on it. Please create new issue.

@ar8001 

> @shajen : if you need access to a remote system with AirSpy Mini .... let me know, and i will send you more donations from time to time to support you and this project

Please follow [this](https://github.com/shajen/rtl-sdr-scanner-cpp/issues/76#issuecomment-1792524376) instruction and send logs.

ryanbytes, 2025-03-28:
Is this project dead?

shajen, 2025-10-16:
No. Checkout newest version https://github.com/shajen/sdr-hub