shajen/rtl-sdr-scanner-cpp #78: Option to use HackRF or RTL-SDR

s3m1s0n1c, 2023-09-18:
Hey Folks,

I can't get this working just with hackrf..

It seems to want to load RTL-SDR as well..

Is it possible to have a option in config to load on or the other..

Or even read whats under devices:
so if hackrf and rtl is there then load both..
but if hackrf is there just load hackrf

Thanks
Sonic

Klammraute, 2023-09-22:
Open Terminal type "hackrf_info"
which firmware version do you use?

crtvrmn, 2024-04-20:
it does. the latest version contains a bug, where this does not work, but when you use the devel version it works just fine.

shajen, 2025-10-20:
Check out the new version https://github.com/shajen/sdr-hub If the bug still exist follow the logs manual and attach them.