shajen/rtl-sdr-scanner-cpp #85: delete all recordings

GarryCZ, 2023-11-06:
Hi, thanks for the great work! I have a question if it is possible to delete all records at once, I have 33274 of them and it is not possible to delete them all at once.

Thank you very much

ar8001, 2023-11-06:
in the denvelop version in admin backend you can select all recordings and delete them in one step

SubBass100, 2023-12-10:
I can report that I also get the Bad Request (400) when trying to delete old data... A cool feature would be to add a data retention setting. You can set it to only keep data so far back in days (or all)     You may only be interested in saving say, the last 30 days worth of data. Every night it deletes everything older than 30 days.