shajen/rtl-sdr-scanner-cpp #86: Dev Branch - Frequency Inaccuracy

vk4tmz, 2023-11-11:
@shajen, 

I've followed the thread / jira issue: [ Development branch #76 ](https://github.com/shajen/rtl-sdr-scanner-cpp/issues/76) and configured 3 RTL-SDR (2xNESDR's and 1xGeneric). All seems to be good with so far however have an couple of issues / request for feature:

**1. Frequency Inaccuracy**

I've noticed while monitoring VHF Maritime 156-158 Mhz, specifically our local Brisbane VTS CH 156.600Mhz, its being reported as slightly off-frequencies (ie 156.610, 156.620 and now 156.640 Mhz).

[image]

I've pulled the docker image version this morning about 2hrs ago so assume I've got latest dev images.

I've been running the master version fine for last 48 hours with same SDR's. 

**2. Audio ticking noise**

I've noticed on the audio recordings it has a ticking noise (i.e. almost like the ignition / spark plug RFI) but the same SDR is fine running under the current master version.  I've attached same transmission (graph, raw and mp3) - https://file.io/X3nfU9Y8PJnD due to GitHub restrictions. 

**3. Feature Request - Specify PPM per device**

Is there plans to allow us to specify PPM per device.  On the master version I could specify PPM which applied to all RTLSDR which is not ideal when only one out of the 3 needs to have the PPM specified as there others are fine.

As always keep up the awesome work!! this tool is fabulous!!!

Mark
(VK4TMZ)

vk4tmz, 2023-11-12:
@shajen,

Update: 

I noticed under master release version that VHF Marine frequencies ending with 5 KHz (ie 156.725 MHz) were being reported  with wrong frequency such as 156.720 MHz and audio was muffled/static.     I eventually nutted out the purpose of the following configuration property and changed it from **10000 Hz** to **5000 Hz** and those frequencies started recording just fine:

**detection::frequency_grouping_size:** 10000 

With this knowledge I located the same / newly renamed property under the "devel" branch release, and updated to 5000 Hz

**other::expert mode::Transmission group size**: 10000

The recordings are now correctly logging against correct frequencies: 156.600 MHz.

Could  I recommend / request that you update the wiki documentation with more details on the config properties ? 

Especially another useful property which was set by default to 2000 ms which prevented a lot of short communications from VTS and vessels that were under 2secs:

Master: **recording::min_time_ms**: 2000
Devel:  **other::expert mode::Transmission min time**: 10000 

I changed my value to suit my needs of 750ms.

Will continue to evaluate the "Devel" release but I'm still being plagued by the humming/buzzing noise.  This occurring on all recorded MP3,  however it should be noted that when I reviewed the raw recorded audio file via GQRX I noticed the buzzing / humming was not there.

Mark
(VK4TMZ)

mikepaxton, 2024-03-09:
> Will continue to evaluate the "Devel" release but I'm still being plagued by the humming/buzzing noise. This occurring on all recorded MP3, however it should be noted that when I reviewed the raw recorded audio file via GQRX I noticed the buzzing / humming was not there.

@vk4tmz did you ever solve the buzzing/clicking sound on the devel branch?  I'm hearing the same and can confirm the .raw file sounds fine in GQRX.  It does seem to happen more with weaker signals.

Klammraute, 2024-03-12:
I encountered the same issue where I observed signals at -5000Hz below the normal frequency.

My solution: adjusting the **sample rate from 2048000 to 2400000** resolved the issue.

Klammraute, 2024-03-13:
Dont try to set the sample rate to 2100000 or something else then 2400000 it will break functionality, if you dont reach you config via the webinterface go to /var/snap/docker/common/var-lib-docker/volumes/sdr_sdr_config and edit it with nano or vi