shajen/rtl-sdr-scanner-cpp #89: Hackrf Error when more than 1 ranges

Klammraute, 2023-11-24:
hi,

When i use more than 1 range on Hackrf i get a error (hackrf_info) tells me that the device is busy and i have to replug it. 
[image]
[image]

When i scan only 1 range on Hackrf all works great. 

I also tried to add it as another device but it still not work and only scan the first range
 :(

shajen, 2023-12-02:
Please ensure the first and second range does not intersect. I suggest you to test new devel version that is much better.

shajen, 2023-12-07:
Follow instructions [here](https://github.com/shajen/rtl-sdr-scanner-cpp/issues/76#issuecomment-1787475909)

shajen, 2025-10-20:
It should work in the new version. Check out the new version https://github.com/shajen/sdr-hub