shajen/rtl-sdr-scanner-cpp #90: Where does SDRScanner store configuration (docker)

SubBass100, 2023-12-18:
Hi, I'm running the Devel branch docker image. I had everything working well but made a mistake and changed my sample rate to an incompatible number which is smaller than the range. I would change it back, however, the SDR options are missing since SDR scanner is now having an error. I've tried to restart the docker image and I've even tried editing the config.json file. Where does SDR Scanner save config changes within the docker image? no matter what I put in the config.json it seems to not read it and goes right back to the wrong settings.  Thanks!

PS. really looking forward to the next release, keep up the good work!

shajen, 2023-12-19:
Hi,
you need to find docker volume named `*_sdr_config`, delete them and restart scanner. `config.json` is no longer valid.

SubBass100, 2023-12-19:
Thank you, for anyone that might stumble upon this post, I found that file folder here:
/var/lib/docker/volumes/sdr_sdr_config/_data
I was able to edit the config.json file within and change my sample rate back to something which was valid.

closed 2023-12-19 by SubBass100