shajen/rtl-sdr-scanner-cpp #91: Django Request Error - Database locked

marcor81, 2023-12-29:
Hello,

I am currently using the dev version and noted that web interface crashes (returning 500 - internal server error).  Can you kindly let me know whether it's an installation issue or otherwise?

sdr-monitor-server-1  | [2023-12-29 13:15:53.117][log            ][django.request][ERROR  ] Internal Server Error: /sdr/transmissions
sdr-monitor-server-1  | Traceback (most recent call last):
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/utils.py", line 89, in _execute
sdr-monitor-server-1  |     return self.cursor.execute(sql, params)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/sqlite3/base.py", line 328, in execute
sdr-monitor-server-1  |     return super().execute(query, params)
sdr-monitor-server-1  | sqlite3.OperationalError: database is locked
sdr-monitor-server-1  | 
sdr-monitor-server-1  | The above exception was the direct cause of the following exception:
sdr-monitor-server-1  | 
sdr-monitor-server-1  | Traceback (most recent call last):
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/core/handlers/exception.py", line 55, in inner
sdr-monitor-server-1  |     response = get_response(request)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/core/handlers/base.py", line 197, in _get_response
sdr-monitor-server-1  |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/contrib/auth/decorators.py", line 23, in _wrapper_view
sdr-monitor-server-1  |     return view_func(request, *args, **kwargs)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/contrib/auth/decorators.py", line 23, in _wrapper_view
sdr-monitor-server-1  |     return view_func(request, *args, **kwargs)
sdr-monitor-server-1  |   File "/usr/src/sdr-panel/sdr/views.py", line 138, in transmissions
sdr-monitor-server-1  |     return page_response(request, "transmissions.html", transmissions, order_by, additional_data=additional_data)
sdr-monitor-server-1  |   File "/usr/src/sdr-panel/monitor/views.py", line 110, in page_response
sdr-monitor-server-1  |     response = render(request, template, data)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/shortcuts.py", line 24, in render
sdr-monitor-server-1  |     content = loader.render_to_string(template_name, context, request, using=using)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/loader.py", line 62, in render_to_string
sdr-monitor-server-1  |     return template.render(context, request)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/backends/django.py", line 61, in render
sdr-monitor-server-1  |     return self.template.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 175, in render
sdr-monitor-server-1  |     return self._render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/test/utils.py", line 112, in instrumented_test_render
sdr-monitor-server-1  |     return self.nodelist.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in render
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in <listcomp>
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 966, in render_annotated
sdr-monitor-server-1  |     return self.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/loader_tags.py", line 157, in render
sdr-monitor-server-1  |     return compiled_parent._render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/test/utils.py", line 112, in instrumented_test_render
sdr-monitor-server-1  |     return self.nodelist.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in render
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in <listcomp>
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 966, in render_annotated
sdr-monitor-server-1  |     return self.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/loader_tags.py", line 63, in render
sdr-monitor-server-1  |     result = block.nodelist.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in render
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 1005, in <listcomp>
sdr-monitor-server-1  |     return SafeString("".join([node.render_annotated(context) for node in self]))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/base.py", line 966, in render_annotated
sdr-monitor-server-1  |     return self.render(context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/template/defaulttags.py", line 194, in render
sdr-monitor-server-1  |     len_values = len(values)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/models/query.py", line 380, in __len__
sdr-monitor-server-1  |     self._fetch_all()
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/models/query.py", line 1881, in _fetch_all
sdr-monitor-server-1  |     self._result_cache = list(self._iterable_class(self))
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/models/query.py", line 91, in __iter__
sdr-monitor-server-1  |     results = compiler.execute_sql(
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/models/sql/compiler.py", line 1562, in execute_sql
sdr-monitor-server-1  |     cursor.execute(sql, params)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/utils.py", line 67, in execute
sdr-monitor-server-1  |     return self._execute_with_wrappers(
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/utils.py", line 80, in _execute_with_wrappers
sdr-monitor-server-1  |     return executor(sql, params, many, context)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/utils.py", line 84, in _execute
sdr-monitor-server-1  |     with self.db.wrap_database_errors:
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/utils.py", line 91, in __exit__
sdr-monitor-server-1  |     raise dj_exc_value.with_traceback(traceback) from exc_value
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/utils.py", line 89, in _execute
sdr-monitor-server-1  |     return self.cursor.execute(sql, params)
sdr-monitor-server-1  |   File "/usr/local/lib/python3.10/dist-packages/django/db/backends/sqlite3/base.py", line 328, in execute
sdr-monitor-server-1  |     return super().execute(query, params)
sdr-monitor-server-1  | django.db.utils.OperationalError: database is locked
sdr-monitor-server-1  | [29/Dec/2023 13:16:40] "GET /sdr/transmissions?reset= HTTP/1.1" 500 5996

Mario

shajen, 2024-01-09:
It looks like problem with your DB inside volume. Try to **stop** all containers, **remove** volumes and try again.

shajen, 2025-10-20:
Check out the new version https://github.com/shajen/sdr-hub