shajen/rtl-sdr-scanner-cpp #92: manifest unknown

panda4us, 2024-01-06:
After pulling the devel branch I am unable to sping up the project with the following error:
`Error response from daemon: manifest for shajen/monitor:latest not found: manifest unknown: manifest unknown`
Please advice

shajen, 2024-01-06:
What hardware platform do you use? amd64, arm32, arm64 or another?

panda4us, 2024-01-06:
Today I've tested it on amd 64 running Ubuntu 22.04.3. Before I had the same issues with amd64 on Windows 10/11.
There is a high chance I am doing something wrong so here are my steps:
1. stop and remove all running docker images
2. git clone
3. git pull origin devel
4. Here I've tried docker-compose pull with the same resuults
5. docker-compose up -d
And thanks for the quick reply!

panda4us, 2024-01-08:
I've checked the docker-compose file, and tried to fix the problem by renaming the image name from "monitor" to "sdr-monitor"
[image]
but now after pulling the images a new error occured:
[image]

shajen, 2024-01-09:
You do something wrong.

Please **stop** all running containers, **remove** all old volumes and run this commands in **fresh** directory:
```
git clone --branch devel https://github.com/shajen/rtl-sdr-scanner-cpp.git .
sed -i 's/latest/devel/' .env
docker compose up -d
```
send full log if it not works correctly.

panda4us, 2024-01-10:
That works... Thanks for helping me out!

closed 2024-01-10 by panda4us