shajen/rtl-sdr-scanner-cpp #94: Docker free Install?

sysadminpower2019, 2024-02-02:
Is it possible to install this without docker? 
Could you provide an idea of the requirements of the project to be installed without docker?

shajen, 2024-02-03:
Yes, follow wiki
https://github.com/shajen/rtl-sdr-scanner-cpp/wiki/Advanced-usage#build-from-sources
but it does not contain web panel and mqtt, you need to setup it separately.

sysadminpower2019, 2024-02-03:
Are there any instructions on how to connect mqtt and web panel to the main
app? Are these required for proper deployment?

> Yes, follow wiki
https://github.com/shajen/rtl-sdr-scanner-cpp/wiki/Advanced-usage#build-from-sources
> but it does not contain web panel and mqtt, you need to setup it
separately.
https://ci6.googleusercontent.com/proxy/Z3TW6Zr3WMTt6fDxvETpRR0O3EN8FGebnPVeQ-vqSGDatgynMGypskoLBRj6F5TrWZnRDVjhBOGRnd7sdMTEziFunPnQ-zLO3ZOBz_GWmxbqFb6g093NKt0_PiwNsHhzTiLLk6mpMOWnbyuAJJpHm4nM2r0u-gXLyymIuG5cfhEP-sMCCA3_aoReRNiWG956ObFIOyts0-DTyMOnag5DvQPNe9ata07lRmaMPSMwzlQhM-sD5g=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABX4XHKSIFINLSQJL43YRXGQTA5CNFSM6AAAAABCXPHNMGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTSX52X6.gif>Message
ID: ***@***.***>

shajen, 2024-02-04:
> Are there any instructions on how to connect mqtt and web panel to the main app? Are these required for proper deployment?

No. 

You have to start any mqtt server manually. 

Then analyze [docker-compose.yml](https://github.com/shajen/rtl-sdr-scanner-cpp/blob/master/docker-compose.yml) how to pass mqtt server arguments to web panel and sdr-scanner.

To start web panel without docker analyze [Dockerfile](https://github.com/shajen/monitor/blob/master/Dockerfile).

Itsisme, 2024-05-23:
Looks like a great tool to use but indeed without Docker.
Anyone got it to work ?
Stuck at 
/usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.4
  warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"

Looks like my distrubution is to far ahead??

./auto_sdr config.json runs ok but to link it to the monitor