shajen/rtl-sdr-scanner-cpp #98: AM FM Modulation

Klammraute, 2024-03-04:
I have a range in which both AM and FM modulation can occur. Now, I initially divided the range into two groups: 26MHz to 28MHz for AM and 26MHz to 28MHz for FM. However, it always picks up AM modulation even though the transmission was in FM modulation. Now, my question is: is it possible for the scanner to automatically choose the modulation, or does it always pick one modulation for a range?
i tried these two settings, but its not working 
[image]

second 
[image]

samsonpwnz, 2024-07-25:
I am also looking to do the same thing, but a different range. I want to force AM modulation in the UHF Air range 225MHz-399MHz. It is scanning this range perfectly, but always demodulates as FM.

avogodro, 2024-08-18:
@Klammraute How did you make those settings?  I too would like to be able to change modulation for certain frequencies.

Klammraute, 2024-08-18:
> @Klammraute How did you make those settings? I too would like to be able to change modulation for certain frequencies.

Which version do you use? master oder devel?

avogodro, 2024-08-18:
I just use latest tag at the end of my image in docker.  shajen/sdr-monitor:latest
I cannot see where the version is listed in the app.