shajen/rtl-sdr-scanner-cpp #109: Config Page Not Showing

WickedxJosh, 2024-05-31:
Hello,

Ive installed the newest version of sdr scanner and am not getting it to show the config page. I installed the dev branch one.

I did it once before on an older Pi and it was showing the config page but no devices. So I wiped that and just did a clean install. Now theres no config page shown.

Any ideas on how to fix this?

ossumpossum, 2024-06-08:
Having the same issue on the devel branch. Config page has one scanner listed for selection, but no devices.  Both soapySDRUtil and auto_sdr correctly recognize the RTL-SDR that's connected. 

~/rtl-sdr-scanner-cpp$ SoapySDRUtil --probe="driver=rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=rtlsdr
[WARNING] SoapyVOLKConverters: no VOLK config file found. Run volk_profile for best performance.
Found Rafael Micro R820T tuner
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=RTLSDR
  hardware=R820T
  index=0
  origin=https://github.com/pothosware/SoapyRTLSDR

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: YES
  Time sources: sw_ticks
  Other Settings:
     * Direct Sampling - RTL-SDR Direct Sampling Mode
       [key=direct_samp, default=0, type=string, options=(0, 1, 2)]
     * Offset Tune - RTL-SDR Offset Tuning Mode
       [key=offset_tune, default=false, type=bool]
     * I/Q Swap - RTL-SDR I/Q Swap Mode
       [key=iq_swap, default=false, type=bool]
     * Digital AGC - RTL-SDR digital AGC Mode
       [key=digital_agc, default=false, type=bool]
     * Test Mode - RTL-SDR Test Mode
       [key=testmode, default=false, type=bool]
     * Bias Tee - RTL-SDR Blog V.3 Bias-Tee Mode
       [key=biastee, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS8, CS16, CF32
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Size - Number of bytes per buffer, multiples of 512 only.
       [key=bufflen, units=bytes, default=262144, type=int]
     * Ring buffers - Number of buffers in the ring.
       [key=buffers, units=buffers, default=15, type=int]
     * Async buffers - Number of async usb buffers (advanced).
       [key=asyncBuffs, units=buffers, default=0, type=int]
  Antennas: RX
  Full gain range: [0, 49.6] dB
    TUNER gain range: [0, 49.6] dB
  Full freq range: [23.999, 1764] MHz
    RF freq range: [24, 1764] MHz
    CORR freq range: [-0.001, 0.001] MHz
  Sample rates: [0.225001, 0.3], [0.900001, 3.2] MSps
  Filter bandwidths: [0, 8] MHz

Klammraute, 2024-08-18:
Most of the time, the settings page fails to appear due to a previous misconfiguration. To fix this, you need to locate the config.json file in the Docker container and identify the error. Common issues include an incorrect sample rate, an extra number, or an error in any statistics. If you can't find the error, replace the configuration with the default one and re-enter all the necessary settings.

At the moment i use the old version of the scanner because of this errors and more.

zero48, 2025-01-14:
> Most of the time, the settings page fails to appear due to a previous misconfiguration. To fix this, you need to locate the config.json file in the Docker container and identify the error. Common [...]

Hello, I have installed the old version ... Still cant make it work . Would you be kind and share your config Json file .Please
'