shajen/rtl-sdr-scanner-cpp #111: DMR Reception?

Cagier, 2024-07-20:
I see from the demos and screenshots that FM and AM are supported.  Will this utility also record digital transmissions like DMR and NXDN.  If so, and there is no actual decoder, is the raw recording captured in a .wav file that could be processed by another program such as DSD+?
Many thanks

RE-Specto, 2025-04-09:
dmr support could be really interesting, considering how widespread it became this days. 
should be pretty simple considering that there are multiple open source decoders to choose from, such as:
https://github.com/f4exb/dsdcc
https://github.com/boatbod/op25
https://github.com/lwvmobile/dsd-fme
and more..

GarryCZ, 2025-04-10:
This project is probably already forgotten, check my repo, I have Client
and server solutions. The client uploads and sends to the server. The
server has a pretty decent web interface. It automatically recognizes DMR
and analog.

https://gitlab.com/globalscanrx