shajen/rtl-sdr-scanner-cpp #115: No config page, how to configure AirSpy Mini then

NLtestuser0001, 2024-08-14:
Hi ya'll,

So, I've got the dev branch running. I can access the page displaying: Hello my dear friend!

But the config page does not load. The admin page does.

How can I setup my Airpsy Mini in rtl-sdr-scanner-cpp without the config page? I am aware of the config.json file but I don't know what to put in it.

Please advise.

Best reagrds.

Klammraute, 2024-08-18:
Most of the time, the settings page fails to appear due to a previous misconfiguration. To fix this, you need to locate the config.json file in the Docker container and identify the error. Common issues include an incorrect sample rate, an extra number, or an error in any statistics. If you can't find the error, replace the configuration with the default one and re-enter all the necessary settings.

At the moment i use the old version of the scanner because of this errors and more.

shajen, 2025-10-29:
Check out the new release https://github.com/shajen/sdr-hub it's much simpler!