shajen/rtl-sdr-scanner-cpp #126: Convert transmission data to .mp3

StiXerrr, 2025-03-05:
Hi, can you tell me how do you convert transmission data to mp3 format? 

[image]

smokedsalmonbagel, 2025-03-06:
I am not the author but I have taken a look at how the app works.  The source code has not been shared yet but is viewable in the docker container.  The app only stores the bin files.  When you go to the endpoint `{base_url}/admin/sdr/transmission/{tid}` the app will take the .bin file and convert it to mp3.  One way to get the mp3 is to scrape it from this location with python.   You can also get the spectrograms in a similar way.

The other approach is to go through the source code in the web app and find the functions that do this internally.  I put the ones you need together here:
https://gist.github.com/smokedsalmonbagel/271c52fa5edca07e8976fa73d37853ab

StiXerrr, 2025-03-06:
Absolutely amazing! It works perfectly! Until now I used a simple decoder in Java, but it was not ideal for lower quality signals. Thanks a lot for sharing your solution!

shajen, 2025-10-29:
All subprojects and new version was published. Details here https://github.com/shajen/sdr-hub