shajen/rtl-sdr-scanner-cpp #127: Compatibility with RTL SDR Blog v4

StiXerrr, 2025-03-11:
I have problem with original RTL SDR Blog v4. After start project I get error: https://pastebin.com/VKPprern
```
sdr-scanner-1         | [2025-03-11 14:01:12.409007] [info   ] [Scanner    ] start thread id: 24
sdr-scanner-1         | [2025-03-11 14:01:12.409039] [debug  ] [RtlSdr     ] check device availability: 0
sdr-scanner-1         | [2025-03-11 14:01:12.485537] [debug  ] [RtlSdr     ] set bandwidth: 002.048 kHz
sdr-scanner-1         | [2025-03-11 14:01:12.548561] [debug  ] [RtlSdr     ] set sample rate: 002.048 kHz
sdr-scanner-1         | [2025-03-11 14:01:12.644015] [debug  ] [RtlSdr     ] set center frequency: 148.715 kHz
sdr-scanner-1         | [2025-03-11 14:01:12.687356] [error  ] [Scanner    ] exception: set center frequency error
```
Any problem with RTL SDR Blog v4?

StiXerrr, 2025-03-12:
It looks like drivers for RTL SDR in master branch is not compatible with original RTL SDR Blog V4.

laurensoosting, 2025-04-17:
I tried the development branch. There I just got it working. The audio quality for the 2 meter band is garbage. 

For the development branch I looked into the reason and found that the default Ubuntu driver "librtlsdr0 (0.6.0-4)" was used. I used the following steps to fix it quick and dirty.

Setup the latest available development environment.
```
git clone --branch devel https://github.com/shajen/rtl-sdr-scanner-cpp.git ~/sdr
cd ~/sdr
sed -i 's/latest/devel/' .env
nano .env # customize settings if you want: timezone, admin account and other
sudo docker-compose up
``` 

Use a second terminal to find the active containers via `sudo docker container ls`. Connected to sdr-scanner container via `sudo docker exec -it sdr-dev_sdr-scanner_1 sh`

Followed the installation instructions from https://www.rtl-sdr.com/v4/

```
apt update
apt install install libusb-1.0-0-dev git cmake pkg-config

cd ~
git clone https://github.com/rtlsdrblog/rtl-sdr-blog
cd rtl-sdr-blog
mkdir build
cd build
cmake ../ -DINSTALL_UDEV_RULES=ON
make
make install
cp ../rtl-sdr.rules /etc/udev/rules.d/
ldconfig
```

Switched to initial terminal and quit application with ctrl+c. Restarted the application via `sudo docker-compose up`

anoduck, 2025-07-10:
Yep, that is pretty dirty. Reminds me of someone I know. 

You could have just forked the repo, created a new branch, and then modified the DockerFile to do the same thing. This way, you would just have to keep your repository in sync with upstream, and your changes would be persistent across builds.

anoduck, 2025-10-29:
@shajen show off 😝 😆